cleanflight/cleanflight #665: PPM Problem

larryho5, 2015-03-24:
[image]
[image]
[image]
[image]

Dear All,

I'm trying to use FrSKY D4R-II on F3 Discovery Board. I'm using A8 Pin and I have set to RX_PPM mode. But It still doesn't work. Does anyone know how to make this work?

frog32, 2015-03-24:
Are you using 1.8.0? I have the same issue using ppm. Using an older version works just fine for me.
There seems to be an issue with ppm in 1.8.0.

tricopterY, 2015-03-24:
The best combo is CF 1.8.0 and Configurator 0.63.0. Configurator may not be reflecting correctly.

hydra, 2015-03-25:
Problem confirmed with 1.7.0 and 1.8.0 - clearly this has been broken for a little while  Thanks for the report.

This is likely also an issue on the sparky board too.
Problem confirmed also on the SPRacingF3.
PPM Works fine on F1 targets so the issue only appears to be related to the F3 targets.

I have mainly been testing F3 targets with Parallel PWM and Serial RX receivers so this PPM issue has slipped though the cracks.

@frog32 do you know which version you last tried when it worked ok?

hydra, 2015-03-25:
The problem occured between 1.0.0 and 1.1.0

https://github.com/cleanflight/cleanflight/compare/v1.0.0...v1.1.0

hydra, 2015-03-25:
@nebbian I have a sneaky feeling the oneshot code might have something to do with it see c0ed21a818e22070507386a503939fc1d775f92f

hydra, 2015-03-25:
@ledvinap Maybe it was this? aa7f5c4a1e66fb838f4ffb95ef1e41038765e1cc

larryho5, 2015-03-25:
yes, I just tryed 1.7.2, 1.7.1, 1.6.0, 1.5.0. and all of this doesn't work on my f3 discovery board

hydra, 2015-03-25:
@nebbian doesn't look like it was the oneshot code.

@ledvinap can you investigate?

hydra, 2015-03-25:
See #113

larryho5, 2015-03-25:
have you tried ppm on F3 Discovery board before? It seems like doesn't work from the version of  1.0.0.

hydra, 2015-03-25:
it works on 1.0.0 here.

larryho5, 2015-03-25:
oh... I just tried many times on 1.0.0 without the configurator, just monitor the output GPIO, it seems doesn't have any change when I move remote stick.

frog32, 2015-03-25:
I see the same problem with a naze32 after flashing and setting ppm using cli. My new job does't allow me to do open source development yet (some paperwork needed), but I will try to bisect the problem and see which commit is causing the problem.

    cleanflight frog32$ git bisect start
    cleanflight frog32$ git bisect good v1.0.0
    cleanflight frog32$ git bisect bad v1.8.0
    Bisecting: 463 revisions left to test after this (roughly 9 steps)

Just 9 steps until you know exactly where the bug has beed introduced.

tricopterY, 2015-03-25:
Just wondering about casting (uint16_t)PPM_TIMER_PERIOD, where  PPM_TIMER_PERIOD = 0x10000
i.e 65536. Shouldn't that be 0xFFFF i.e 65535.
Then we don't have to do:
TIM_TimeBaseStructure.TIM_Period = PPM_TIMER_PERIOD - 1;

thenickdude, 2015-03-25:
@frog32: PPM definitely works on the Naze32 with 1.8.0, that's what I fly every day.

frog32, 2015-03-25:
@thenickdude I have tried with 3 different boards (naze32 rev4, rev5, rev5 acro) and all did not work. I will try again this evening and try to make this reproducible.

thenickdude, 2015-03-25:
It definitely works on Naze32/F1... otherwise since PWM is broken on 1.8.0, the only people who would have been able to fly 1.8.0 would be using Serial Rx and the issue tracker would be exploding from tickets.

What receiver are you using, and have you added the appropriate jumper (if required) to enable PPM output on it?

ledvinap, 2015-03-25:
@hydra: Will get to in ASAP. Can you please check if Sparky is affected? Sparky is only 303 HW I can test ...

hydra, 2015-03-25:
@ledvinap Interestingly the Sparky PPM is fine with current master.

I will be shipping a SPRacingF3 board to you when they roll off the production line.  I can use some project donations and order you an STM32F3Discovery and have it shipped to you if needed.

ledvinap, 2015-03-25:
TIM16 and IRQ sharing may be the difference. But on SPRacingF3 TIM2 is used for PPM.

Generated code looks OK ... 

Can you try commenting out TIM16 on SPRacingF3 ?

hydra, 2015-03-25:
@ledvinap yes i can try that on the SPRacingF3 target.  Any ideas on what to try on the STM32F3Discovery?

@ledvinap Not sure if this is related. but the pwmEdgeOverflow callback (yes, pwm) never gets call on the SPRacingF3 for the first 4 PWM inputs when powering off the RX.  I can raise a separate issue for this if you think it's unrelated.

hydra, 2015-03-25:
@ledvinap Ok, i disabled `TIM16` in the SPRacingF3 target and that does not change things.  PPM still dead.

If i set a breakpoint in the `ppmOverflowCallback` it's only ever called once, the first time it overflows.  It's never called again and thus `ppmDev.largeCounter` is never increased. (remains at 65536).

This then means that `ppmDev.stableFramesSeenCount = 0;` is eventually executed resulting in non-working PPM.

Perhaps my thought above that `pwmEdgeOverflow` also not being called for `TIM2` on the SPRacingF3 is somehow related?

hydra, 2015-03-25:
Ok, i have found the issue.  It's in the configuration of the ARR register.  Using timCCR_t instead of uint16_t fixes it.  Doing more testing.

frog32, 2015-03-25:
ok I checked again and naze32 seems to work, but the gui is unusable and gets a lot of ```MSP data request timed-out: 1```warnings. But looking at the motors page shows stuff should be working.

hydra, 2015-03-25:
@frog32 I think you might be using latest firmware with old configurator.  Use 1.8.0 with 0.63.x.

@ledvinap I spoke to soon, while the fix works for TIM2 on the SPRacingF3 it does not fix the issue on the STM32F3Discovery.

ledvinap, 2015-03-25:
@hydra: Can you elaborate a bit, please? I don't see `ARR` access except 'timCCxHandler' and 'TIM_TimeBaseInit', both seem to be typed correctly..

hydra, 2015-03-25:
@ledvinap I'll show you a branch that fixes it on both targets and will comment on the commits, give a few mins.

tricopterY, 2015-03-25:
I may sound stupid - I just do not understand #define PPM_TIMER_PERIOD 0x10000. I thought the 16bit timer would be 65535 i.e 0xFFFF.

hydra, 2015-03-25:
@tricopterY there's a -1 floating around in the code where the timer ARR is set.  Probably needs some cleanup.

hydra, 2015-03-25:
@ledvinap @tricopterY Here's a workaround commit.

https://github.com/cleanflight/cleanflight/commit/b85351d250c22f387d090cdb0ddc7c447521976f

It's definately not the final solution.

ledvinap, 2015-03-25:
@tricopter: That was my decision. 
STM32 timers have period ARR+1. So if you want maximum period ( 0x10000 ticks), you load 0xffff into register. Old code got this wrong, if was ignoring the +1 part. Not too big error, but wrong.

New `timerConfigure()` takes timer period as `period` parameter, with 0x10000 passed as zero (this is single period value when 16bit is not enough).

hydra, 2015-03-25:
TIM2 on SMT32F303 has a 32bit ARR.

frog32, 2015-03-25:
Still got issues, everything works just fine but when I connect a ppm receiver I get MSP request timeouts and crc errors using development branch of configurator https://github.com/cleanflight/cleanflight-configurator/commit/52d76709777eb044fd89e3da128b9782b3ea0eb4 and v1.8.0 of cleanflight.

hydra, 2015-03-25:
@frog32 I think your problem is different then, please open a new issue and full describe your symptoms, hardware, wiring, rx/tx details and configuration dump and steps to repeat.

ledvinap, 2015-03-25:
con you try to change https://github.com/cleanflight/cleanflight/commit/b85351d250c22f387d090cdb0ddc7c447521976f#diff-f7c531a600c2ba76bf2540035c446c6dR369
```
TIM_TimeBaseStructure.TIM_Period = (period - 1) & 0xffff; 
```

hydra, 2015-03-25:
Yes, after the above comments it seems like it might be that simple. 1 sec.

hydra, 2015-03-25:
@ledvinap that fixes the SPRacingF3 target, but the STM32F3Discovery/Chebuzz is still failing.

hydra, 2015-03-25:
[image]

Here's a debug session showing the issue.  Note the values of the deltaTime 7 and 8.

ledvinap, 2015-03-25:
Both use TIM1 for PPM and have TIM16 used elsewhere. So it may be TIM1_UP_TIM16_IRQn.

If TIM16 is not initialized, TIM1_UP_TIM16_IRQn may stay at default priority and thus interrupt Edge handler. It is race to solve, but probably not critical.

deltaTimes[7] = 0xffff0000 + 991 ... like if largeCounter was decremented 
deltaTimes[8] - capture = 0xfeef ... not sure if thisis significant

ledvinap, 2015-03-25:
There is IMO no way to handle large counter correctly without greatly complicating the code. Timer overflow and capture interrupt may arrive in any order.

PPM should work quite fine with 16bit timer delta (oneshot 8ms overflow must be checked), overflow ISR may be used only to check for timeouts.

I'll look into it tomorrow ...

frog32, 2015-03-25:
sorry my problem seemed to be just an emf issue, used uhf receiver with telemetry using just a monopole antenna, this uses then the cable as counter part which introduced a lot of noise. only real problem is that cleanflight locks up when receiving noisy ppm stream.

hydra, 2015-03-25:
@frog32 this is expected behaviour since the PPM handlers will be continually called by the MCU because the hardware is configured to run then when the signal transitions to/from high/low.  The noise will then result in the handlers being called continually and will prevent the CPU from handling other tasks.  Fix the noise, problem solved. :smile:

hydra, 2015-03-25:
@ledvinap great, thanks for your valuable input. Look forward to seeing your solution.

ledvinap, 2015-03-26:
@hydra: the ppm noise problem may be partially solved by using long enough input filtering, so that there is enough time to process IRQs. Setting filter time to minimum expected pulse width should do the trick (and does not degrade performance).
Another useful improvement is to disable IRQ for some time if too short pulse is detected (or invalid PPM frame is received).

`Fix noise` is only partial remedy - it is not OK when unexpected conditions can kill Cleanflight completely.

thenickdude, 2015-03-26:
But if you have that much noise on the input, can you really have reliable control? It seems like slowing down the FC is a best-case scenario when it could easily just take off with no intentional input.

hydra, 2015-03-26:
See #676

ledvinap, 2015-03-26:
#676 should fix this ... 

TIMx_UP IRQs  were not enabled, so the flag got processed much later when compare IRQ occured.

Old code did not use overflow  on TIM1 to hide another bug, and 16bit capture differences usually worked (when there was no overflow in PPM frame)

The ppm code is still buggy, but should work reasonably well now .. 

@hydra: Can you test it? I don't have access to PPM receiver ...

hydra, 2015-03-26:
@ledvinap yes, ignoring noise at the FC would be good

@frog32 please open a new issue and mark it as enhancement.

It's also not obvious but if you're only sending telemetry when armed and the noise from telemetry transmissions cause problems then you may never notice the problem in the configurator when using a shared msp/telemetry port.

hydra, 2015-03-26:
@ledvinap yes, i might be able to test this a bit later tonight.

I'm going to be very busy over the next two days so might not get the chance till sunday.

If others wants to test on whatever targets they have handy that would be great - please provide feedback here.

@ledvinap what other changes do you think need to happen in the PPM code to avoid the race conditions you mention, can you give me details on when the race conditions now occur?

ledvinap, 2015-03-26:
@hydra: As state above, overflow and compare IRQ may occur in 'wrong' order.
(`ARR=99, largeCounter=0`)
when Up has priority:
```
CC1 = 98
OVERFLOW
UPhandler: largeCounter+=100
EdgeHandler: largeCounter + CC1 == 198
```
when CC has priority:
```
OVERFLOW
CC1 = 1
CCHandler: largeCounter + CC1 == 1
UPHandler:   largeCounter+=100
```

You can handle this by assuming that handler delay is bounded, checking if CC register is close to overflow (and from what side) and checking UIF flag. It will get a more complicated when ONESHOT is in effect, but should be doable.

hydra, 2015-03-26:
Ok, the fix in PR #676 works fine on the PORT103R, SPRacingF3 and the STM32F3Discovery/ChebuzzF3.

I will merge that now and perhaps you could look into handling the race condition so that frames are not lost?

ledvinap, 2015-03-26:
The code has this problem for long time and nobody noticed. I'am slowly working on new IO implementation and other improvements, I'll look at PPM in the process.

I would like to split PPM handling into input queue and data processing (similar to softserial), so that high priority timer ISR is short and without spikes. This is necessary for reliable softserial.

Also oneshot may be probably implemented as 'sofware generated' (setting CCRx before each transition, like softserial output). That will remove timer skipping and will improve PPM input/softserial and oneshot timer sharing...

digitalentity, 2015-03-26:
Why use a timer for PPM/PWM? The same functionality can be easily
implemented using EXTI with probably even less software processing.
Moreover, we will be able to use any hardware pin for pwm/ppm receiver
giving us more flexibility and freeing up a timer for other tasks.

ledvinap, 2015-03-26:
@digitalentity : Using exti will add interrupt latency to PPM measurement. Current interrupt latency is ~10us in some cases, this will create ~10 counts PPM noise.
 
Using EXTI for PWM won't work for some combinations (EXTI multiplexer) and accuracy problem will be the same ...

digitalentity, 2015-03-26:
@ledvinap, my bad, forgot about latency.

hydra, 2015-03-30:
I will close this, we have working code which can be further improved in due course.

closed 2015-03-30 by hydra