cleanflight/cleanflight #671: Obsolete docs for softserial

xoxota99, 2015-03-25:
Setting serial_port_xx_scenario appears to no longer exist in 1.8. Configurator "ports" tab appears all jacked up (showing "UART1" multiple times, and nothing else, on Naze32). See attached.
[image]

thenickdude, 2015-03-25:
You're using the wrong Configurator version, please use the 0.63.0 version mentioned as required in the release notes:

https://github.com/cleanflight/cleanflight-configurator/tree/development

closed 2015-03-25 by thenickdude

xoxota99, 2015-03-25:
Thanks! Doesn't configurator update automatically? How can I tell what version I'm using?

xoxota99, 2015-03-25:
Uninstalled and redownloaded the latest version from Chrome web store. No change.

digitalentity, 2015-03-25:
Please read release notes to 1.8.0 release here: https://github.com/cleanflight/cleanflight/releases/tag/v1.8.0
You cannot use the Configurator from the Chrome Web Store, you should manually install configurator from "development" branch here: https://github.com/cleanflight/cleanflight-configurator/tree/development

You install it this way:
1) Download it as zip, unpack to some directory (for example D:/cleanflight-configurator)
2) Start Chromium or Google Chrome and go to tools -> extension
3) Check the "Developer mode" checkbox
4) Click on load unpacked extension and point it to the Cleanflight Configurator directory (for example D:/cleanflight-configurator)

hydra, 2015-03-25:
Yes, all that and please search before opening new issues!