cleanflight/cleanflight #672: "Save and Reboot" from Configuration tab doesn't do either of those things on Naze32.

xoxota99, 2015-03-25:
Repro:
1) Connect to Naze32 (running Cleanflight/NAZE 1.8.0 Mar 20 2015 / 18:39:17 (730d0fe))
2) Go to "Configuration" tab.
3) Change any setting, e.g. set "Yaw Adjustment" to -90. 
4) Click "Save and Reboot". Observe the board does not actually reboot.
5) Disconnect the board and reconnect. Observe the change to "Yaw Adjustment" has not actually been saved. Same prblem occurs for all settings, on all tabs. Only place settings can actually be saved reliably is through the CLI tab.

digitalentity, 2015-03-25:
Make sure you are using 0.63 version of the configurator. "Save and Reboot" is apparently working on my Flip32 (Naze32 clone) with 1.8.0 cleanflight and 0.63 configurator.

xoxota99, 2015-03-25:
Thanks. Where can I get latest version? I uninstalled and reinstalled from Chrome web store, and no difference (that appears to be v 0.62.1?)

xoxota99, 2015-03-25:
Resolving, as this is a configurator issue. This is the wrong issues repo.

closed 2015-03-25 by xoxota99