cleanflight/cleanflight #677: motors spin up after an FC reset

ummagawd, 2015-03-26:
on 1.7.2 with Immersion's Vortex quad, oneshot enabled

I noticed that anything that causes the FC to restart, causes the motors to spin up for a brief second.

Kinda scary when you're not expecting it.

Anthony told me he spoke to you (Dominic) about it, but I didn't see it so I just wanted to make sure it wasn't missed. My apologies if it was already addressed and I just didn't see it.

stronnag, 2015-03-26:
smells like yet another instance of #123

ummagawd, 2015-03-26:
thanks stronnag, i missed that issue. I believe i'm reporting the same thing here

I will go ahead and close this ticket :) and watch that one instead

closed 2015-03-26 by ummagawd

stronnag, 2015-03-26:
Please leave a comment on #123, especially details of the ESCs and motors.