cleanflight/cleanflight #678: Blackbox strange axisI data

muellerr, 2015-03-26:
I'm using Cleanflight 1.8 and a naze32 board on one of my quadcopters. Thanks to the newest version of Cleanflight I can now log flight data to the onboard memory.
Most of the data makes sense. What does not make sense is the "axisI" that does not change over very long periods of time. See the following image:
[image]
What could be the cause of the problem?

thenickdude, 2015-03-26:
That's normal, the I term can only be reduced when the error changes sign, which will either happen when you change direction, when you're hovering, or if the I term is correcting for an offset CoG it might never drop.

thenickdude, 2015-03-26:
(the I term saturates due to limiters which is why it flattens out like that. If there were no limiters you'd probably see it moving around freely instead of sitting at the saturation point)

ledvinap, 2015-03-26:
@muellerr : What PID are you using?

muellerr, 2015-03-26:
@thenickdude: thanks for your answer. There seem to be at least two different levels present. Do these limits change depending on flight mode?

@ledvindap: On the axis shown I use P=4, I=0.025, D=25. But it seems far from the optimum.

ledvinap, 2015-03-26:
@muellerr : And what PIS are you using? (`set pid_controller=n`)

muellerr, 2015-03-26:
I'm using 0.

closed 2015-05-07 by thenickdude