cleanflight/cleanflight #680: Oneshot125 Mode results in insufficent throttle control in 3D mode.

SystemsGuy, 2015-03-28:
When using SimonK escs configured for reverse mode - RC_PULS_REVERSE = 1, and using Oneshot125, CF throttle ranges appear "off".  Full max throttle results in 1/3 of full motor power, and full min throttle results in a equal (1/3) amount of reverse thrust.  The deadband range is also very large.

Disabling Oneshot125 in CF causes the motors to react correctly - small center deadband, and full positive / reverse thrust at the extremes.

Of note is the fact that when you enable RC_PULS_REVERSE, it is recommended that you disable stick calibrate (RC_CALIBRATION = 0) and that you flash a fixed calibration to the esc's eeprom.

thenickdude, 2015-03-28:
I hooked a logic analyser up to a motor pin and tested it on the motors tab with mincommand = 1000, minthrottle = 1100, maxthrottle = 1900, ONESHOT125, but without 3D, with these results:

```
Motor command 1000: Pulse length 125us
Motor command 1100: Pulse length 138us
Motor command 1900: Pulse length 238us
```

Then I enabled 3D and measured again:

```
Motor command 1000: Pulse length 125us
Motor command 1100: Pulse length 138us
Motor command 1460 (neutral3d): Pulse length 183us
Motor command 1900: Pulse length 238us
```

So pulse length generation under OneShot is the same between using 3D and not using 3D, and pulse times look perfectly to spec.

I moved my throttle around on my Tx and confirmed that on the motors tab, the FC's motor commands spanned the full range from mincommand at 1000 to maxthrottle at 1900.

I think your ESC is probably just not calibrated for the right pulse length.

SystemsGuy, 2015-03-29:
Hey Nick - tried a reflash with sticks calibration allowed, and used the motors commands to set max, min, and neutral with the beeps as expected.  Unfortunately exact same results, so I've opened an issue over at tgy! https://github.com/sim-/tgy/issues/75

thenickdude, 2015-05-07:
SimonK now has a fix for this in its trunk.

SystemsGuy, 2015-05-08:
Tested and seems to be working fine!

closed 2015-05-08 by SystemsGuy

lawrence4, 2015-06-16:
After I set the PID_CONTROLLER=1,
meet the same power down problems.
But I change back the PID_CONTROLLER=0,
the problems is not more.