cleanflight/cleanflight #681: Please enable Multiplex SRLX sbus

Fly3Fan, 2015-03-29:
Could somebody please implement SRXL to be used in CLEANFLIGHT. Otherwise MULTIPLEX users cannot utilise all the nice CF features that require serial signal.

Here in Germany Multiplex is very popular but still SRXL is only popular in the helicopter scene so far (Microbeast e.g. have implemented it).

Would be great if someone could help!

Here you can find the documentation of Multiplex SRXL ("SRXL-MULTIPLEX_V2.pdf" first two pages in German language, page 3,4 the same text/specification in English. Same in "Multiplex-Sensor-Bus_V2.pdf", English starts at page 11 ).
http://www.multiplex-rc.de/en/service/downloads/interface-descriptions.html

This is the MULTIPLEX equipment (used in Germany a lot):
Sample receiver documentation at:
 http://www.multiplex-rc.de/en/service/downloads/manuals.html?eID=dam_frontend_push&docID=2729
These transmitters are being used:
http://www.multiplex-rc.de/en/products/categories/products/details/productgroup/aircraft-models/productcategory/imperator/product/profi-tx12-m-link-einzelsender-24-ghz.html?tx_lwshopitems_pi1%5Baction%5D=show&tx_lwshopitems_pi1%5Bcontroller%5D=Product&cHash=36185909139513f4201fed2118c5d42f
and
http://www.multiplex-rc.de/en/products/categories/products/details/productgroup/aircraft-models/productcategory/royal-pro-m-link/product/royal-sx-action-16-kanal-telemetrie-set.html?tx_lwshopitems_pi1%5Baction%5D=show&tx_lwshopitems_pi1%5Bcontroller%5D=Product&cHash=1c7929b505f8396c175183676da65174

tracernz, 2015-03-29:
Try X-Bus mode B. I've heard it's the same protocol by a different name.

http://www.multiplex-rc.de/en/service/downloads/interface-descriptions.html

 http://www.multiplex-rc.de/en/service/downloads/manuals.html?eID=dam_frontend_push&docID=2729
http://www.multiplex-rc.de/en/products/categories/products/details/productgroup/aircraft-models/productcategory/imperator/product/profi-tx12-m-link-einzelsender-24-ghz.html?tx_lwshopitems_pi1%5Baction%5D=show&tx_lwshopitems_pi1%5Bcontroller%5D=Product&cHash=36185909139513f4201fed2118c5d42f
http://www.multiplex-rc.de/en/products/categories/products/details/productgroup/aircraft-models/productcategory/royal-pro-m-link/product/royal-sx-action-16-kanal-telemetrie-set.html?tx_lwshopitems_pi1%5Baction%5D=show&tx_lwshopitems_pi1%5Bcontroller%5D=Product&cHash=1c7929b505f8396c175183676da65174

---
https://github.com/cleanflight/cleanflight/issues/681

hydra, 2015-03-30:
@Fly3Fan please can you confirm whether existing X-Bus support works for you?

Fly3Fan, 2015-03-30:
"configuration" tab:  
   'Receiver Mode': RX-PPM enabled
   'Serial Receiver Provider': XBUS_Mode_B ticked (-> blue background)
nothing marked under "Other Features"
-> safed, rebooting, ready

now select "Receiver" tab
straight line appears at ~1600 and moves from right to left 
if I now connect the Lipo, the movement becomes stuttering and moves about 3 to 4 steps per sec. to the left.
all coloured indicators for TX input stay at 1500, no TX-stick movement visible
if I try to change to an other tab "waiting for data" appears on the screen. Unable to leave this tab other than disconnecting Lipo from copter.
Same with 'Serial Receiver Provider': XBUS_Mode_B_RJ01
So obviously SRXL does not seem to be implemented.

Fly3Fan, 2015-03-30:
As far as I understand from a friend the header of SRXL seems to be the same as JR but the set of data which is transmitted varies and thus checksum also varies/is wrong; as can be seen from the MULTIPLEX SRXL documentation linked in my first post. Hope that is not to confusing info from a 'non-programmer-fool' ;-)  
Thanks in advice for your help! 
Stefan

tracernz, 2015-03-30:
Ahh, [this RC Groups post](http://www.rcgroups.com/forums/showpost.php?p=26393376&postcount=4) sheds some light. The [BeastX (helicopter flybarless controller) wiki has some info too](http://wiki.beastx.com/index.php/SRXL_-_Serial_Receiver_Link_protocol).

Fly3Fan, 2015-04-13:
Unfortuantely Multiplex is not willing to contribute here (see eMail below).
":..die Beschreibung des SRXL können Sie von der Multiplex Homepage herunterladen. Für einen Open Source Programmierer sollte das kein Problem darstellen. Bei uns im Haus ist jedoch keine Kapazität für Fremdprojekte vorhanden.

Mit freundlichen Grüßen,
Ihr MULTIPLEX-Team"

Anybody here who would be so kind and take care of this issue to implement SRXL?

ledvinap, 2015-04-13:
Should be quite easy. Anyone in CZ/Pilsen with SRXL equipment?

Fly3Fan, 2015-04-16:
Hi, can I do anything to support this? (sorry, I am not a programmer)

 

Regards

Stefan

Paulraaf, 2015-04-26:
FYI:
I have my naze32 running perfectly well with a Multiplex RX-5 M-Link receiver. I'm running cleanflight 1.8.1 with X-bus selected. I needed to enable serial output in my receiver though.

Fly3Fan, 2015-04-27:
Hi,That sounds fantastic. Could you please send screenshots of your settings in the gui?
Thanks in adviceStefan

GrußStefan Zisler- - - - - - - - - - - Kurz und knapp vom Smartphone gesendet.

Paulraaf, 2015-04-27:
[image]

[image]

[image]

Here they are.
This is what I had to do:
- Enable serial output in my receiver with the multimat programmer (also possible with a pc if I'm correct)
- Set FastResponse to OFF in the modulation menu of my transmitter (Multiplex Royal Pro 9)
- Enable SerialRx checkbox for UART2 in the ports tab of cleanflight
- Select SerialRx and XBus mode B in the configuration tab.

Note: I'm using the RX-5 M-link receiver so only the first 5 channels work.

Note2: I do think that not all Multiplex receivers might work. The documentation for SRXL mentions a 12 and 16 channel version so apparently my receiver puts out the 12 channel version of the protocol because that is the same as current implementation of xbus mode B.

Fly3Fan, 2015-04-27:
Hi,

Thanks for the pics! That really looks interesting and I am looking forward to test this with my RX-9 M link (hopefully this week, not sure if I’ll find time for that).

I’ll keep you informed. Would be great if it works…

 

Regards

Stefan

 

 <https://cloud.githubusercontent.com/assets/12120879/7343611/b46dfeb6-ecc3-11e4-9881-25edbecee197.png> 

 <https://cloud.githubusercontent.com/assets/12120879/7343612/bad78e16-ecc3-11e4-9cbb-dc262eca918e.png> 

 <https://cloud.githubusercontent.com/assets/12120879/7343615/c45441b4-ecc3-11e4-90bc-607d6803d902.png>

terminar, 2015-05-02:
Hi,
i am in contact with Multiplex since some time and this weekend i have received test equipment to help supporting Multiplex RC in cleanflight. This means of course testing SRXL (XBus Mode B is "based" on SRXL meaning it's the same), due to a first test today i can confirm it's also working with the RX-16-DR pro M-LINK receiver (currently only with 7 channels mapped). 
I am also trying to check if i can add support for MSB meaning telemetry support.

Due to the fact that i am doing this im my spare time and i have never dived into the protocols of Multiplex before i can't give you a timeline but i'm on it.

I'll give feedback the next days and weeks. You can also find me nearly daily on freenode at #cleanflight channel.
Bye
Björn

Fly3Fan, 2015-05-11:
Hi,
just tested with RX-7 M-Link and  RX-9 DR M-Link (both software updated + SRXL enabled).
Both not working via SRXL on naze32 :-(

drone-zone, 2015-08-18:
First of all I want to thank Paulraaf for his great summary of needed steps to get XBUS going, that confirms what I have been trying lately. Also a big thank you to everyone else participated in the implementation of X-BUS. Great job!

Here comes my question, though: I am trying to get a Jeti Receiver talk to Cleanflight over the UDI protrocol (which equals X-BUS Mode B). I am using the latest CF version together with a CC3D board. I have tried the described steps above. Nevertheless, no luck so far. Am I assuming correctly that the signal has to go in to PIN 5 on the RX-Panel having Soft-Serial activated on the CC3D? Or do I have to use Flexi or Main-Port? 
Another connected questions would be: Will there still be OneShot125 support when using a serial receiver on CC3D? In OpenPilot software (GCS) it reads as if OneShot125 only would work when using PPM input on RX Pin 8. I could not figure out yet, if that's an limitation of OpenPilot software or the board itself.

Appreciate your input! :)

paulboes73, 2015-09-08:
Everyone I'm having similar problems getting xbus in mode b working, and my radio is a JR XG14 with xbus set to mode "B" in the transmitter.  Firmware on the Naze32 board is 1.9 and running CleanFlight.  I've tired to set it through the GUI and through the CLI and it doesn't seem to work for me.  All I get is dead sticks on the receiver tab.  Now I use xbus mode b on all my heli flybarless gyro systems with no issues, but I can't seem to get it working on my Naze32 board.  Any help would be appreciated.

paulboes73, 2015-09-08:
I have xbus mode b working in my JR XG14, but there are channels that are wrong.  When I move my throttle channel Aux1 moves.  When I move Yaw my Pitch channel moves.  Can you change the channels manual in Clightflight.  The channel map is set to TAER1234 which is for JR, Spektrum, and Graupner.

paulboes73, 2015-09-08:
Problem resolved now and everything is working on my JR XBUS receiver.  The final step I had to do is go into CLI and use the map command to map the channels for JR.

Command Line: map AER1T234
This fixed my problem with the incorrect channels for JR.

Fly3Fan, 2016-01-05:
Hi,
I succeeded with SRXL via PIN4 (using an Skyline32 Mini) and RX-7-DR MLINK).
Now I would like to add an OSD but there is no RX/TX separated on this Mini board. How can I configure that via UART or SOFTSERIAL? Pleas help!

closed 2017-02-25 by hydra

Juan-ma, 2017-04-14:
Hello,

I have a Revo board with BTFL 3.1.7 and a Multiplex RX5 M-Link configured as SRXL receiver. I have configured Port 2 (UART1) as "Serial Receiver" and in the Configuration section, Receiver as "Serial-based receiver". In the "Serial receiver provider" I have tryed both SRXL and XBUS_MODE_B with no success. With XBUS_MODE_B provider selected, I see all chanels fixed at 1500 in the receiver section. With SRXL provider is selected, when I move right stick all channels move randomly.
I have tryed with older versions of betaflight even with cleanflight but the same thing happens. I have sniffed the serial output of the receiver and it's OK accord the Multiplex description of the protocol. I have tryed with other serial ports of the Revo with the same result.
I have had a look to the code and it seems that I should select "XBUS_MODE_B" for the Multiplex receiver and SRXL seems to be used with Spektrum receivers:

https://github.com/betaflight/betaflight/blob/master/src/main/rx/xbus.c#L62
https://github.com/betaflight/betaflight/blob/master/src/main/rx/rx.c#L219

Maybe the problem is the baudrate of the serial port in the revo (multiplex receiver is 115200).