cleanflight/cleanflight #684: [enhancement] Add flight mode and failsafe state logging to blackbox

digitalentity, 2015-03-30:
This feature will further improve ability to investigate flight issues. I personally was trying to figure out why gps poshold does not work. Blackbox log was showing 9 sats in view, mode switch was configured correctly, but gps poshold mode simply didn't work. It took me a while to figure out my GPS receiver was buggy - didn't report a 3d fix and poshold mode didn't get enabled. If blackbox was logging active flight mode, investigating my issue would took much less time.

payne92, 2015-05-02:
+1 for this request

We're trying to debug a problem now, possibly related to a flight mode issue.

nathantsoi, 2015-05-25:
Agreed, lmk if I can help get this pulled in.

PR #907 (2015-06-30): Blackbox - Add logging for flight state, flight mode, and failsafe state
Fixes #684 

Due to code improvements, code size shrinks by 50 bytes with this addition.
closed 2015-06-30 by thenickdude