cleanflight/cleanflight #685: Blackbox current logging incompatible with virtual current meter

digitalentity, 2015-03-30:
When using a virtual current meter blackbox records invalid data.
Currently blackbox is using amperageLatestADC, which is not getting updated when using virtual current meter.

digitalentity, 2015-03-30:
The easiest way to get things working whould be to modify virtual current sensor code to update amperageLatestADC with some 'virtual' ADC data. @thenickdude, what do you think?

digitalentity, 2015-03-30:
The other and more corrent way would be to record actual 'amperage' variable, however this would require changes not only to cleanflight, but also to blackbox-tools and blackbox-log-viewer.

thenickdude, 2015-03-30:
The information content provided by the virtual current sensor is zero so it doesn't seem like it's worthwhile to log it in the first place? It can be simulated after the fact on a blackbox log from the throttle data.

digitalentity, 2015-03-30:
Virtual current sensor provides a reference. It is not unrelated to actual current being consumed by aircraft. I was surprised when I saw 0 mAh consumed during a 5 second motor test in my blackbox log.
BTW, enabling current sensor (virtual or not) also enables logging current data to blackbox. As we are logging this anyway, why not log something that is more useful than just a bunch of zeros.

ledvinap, 2015-03-30:
Disabling current logging when virtual sensor is in use is IMO good fix.
Virtual current sensor parameters can be saved in log header and data recreated in similar way to '--simulate-imu'

digitalentity, 2015-03-30:
@ledvinap good solution. Will save bandwidth and space for blackbox log, yet will allow to see simulated current data in blackbox logs.

thenickdude, 2015-03-30:
@digitalentity the information content of the virtual current sensor (from an information-theory perspective) is zero, there's no point in logging fields that are entirely derived from other logged fields as it just adds overhead to the log.

The virtual current sensor doesn't cause any overhead in P frames, as the delta is always zero and it's packed into a set of fields where zero deltas don't cause any additional bytes to be recorded. However, it does add two bytes to "I" frames. I'll update the logging condition to exclude the virtual current sensor.

Virtual current sensor parameters are actually recorded already (as they share the `current_meter_scale` and `current_meter_offset` settings used by the ADC).

closed 2015-03-30 by thenickdude

digitalentity, 2015-03-30:
@thenickdude, thanks for fixing this.

commit 7940ac20 (2015-03-30): Blackbox: Don't log amperageADC when current meter is virtual

Closes #685