cleanflight/cleanflight #687: ESC passthrough for flashing?

dkisselev, 2015-03-30:
Just curious if anyone's considered implementing an ESC passthrough mode,

For those ESC's that have servo connector flashable firmware it'd be awesome if CF could act as a linker so you can change blheli settings, update firmware, etc.

Biggest roadblocks I could see are: 

a) Handing off the serial port might be an issue (the configurator would need to send a "passthrough to motor 1" command, and disconnect, then when the FC is powercycled it would go back to configurator mode).

b) Flash space, not sure how much we have right now but it might be wasteful for a feature that's not directly related to flight.

Coolest benefit is that it saves you from having to take your quad apart to get at those connectors when trying to tweak things.

ledvinap, 2015-03-30:
Is there some programming protocol documentation?

dkisselev, 2015-03-30:
Nothing explicit that I know of, but it's been implemented as an Arduino sketch: https://github.com/c---/ArduinoUSBLinker/blob/master/ArduinoUSBLinker.ino

That's what I used to flash my Afro ESC's and it worked great. Basically takes serial input and bitbangs the right sequence out.

hydra, 2015-03-30:
Duplicate of #216

closed 2015-03-30 by hydra