cleanflight/cleanflight #691: Add support for rotor arm tilting mechanism

susheelpuranik, 2015-03-31:
Hi all,

We have already added dynamic tilting mechanism to the arms of a multirotor by using a single servo motor and a separate Arduino board. The tilting mechanism adds vector thrust do the whole system and the flight and speed tests we've done confirmed an increased speed and better FPV camera view by keeping the chassis horizontal, among other advantages. Check this video out of the maiden tests with the basic code: https://youtu.be/jq7iHHFluAI 

More specifically, we are currently using an Arduino Nano board in between the flight controller and the receiver which provides certain PWM to both the flight controller (FC) and the tilt servo. For testing stage we have written a logic in Arduino as follows:

- Two inputs to Arduino board: ELEV and AUX1 from receiver (Rx)
- Two outputs from Arduino board: PWM to tilt servo and to ELEV channel of the FC

The tilt system currently has two modes:

1. Dynamic tilting: the tilt servo is actuated dynamically depending on the ELEV channel and the FC is also receiving a certain PWM.

2. Fixed arms tilt: the user sets a certain tilt angle to the arms and the FC receives the same ELEV signal than the Rx outputs.

Both modes are now handled with a single potentiometer (POT) aux channel in the RC transmitter (Tx): for the POT below 1500 us, the mode is 1 (Dynamic tilting), otherwise the POT sets a certain angle to the arms and the Arduino outputs the same PWM to the FC as received by the Rx.

This product has been already tested and published and given the interest and performance advantages of a tilt mechanism in multirotors, we think that it could be included in Cleanflight (CF). Below our opinion of how this could work in CF and the first set of parameters that we could have:

To implement the system as it is now (open to improvements and other features) we would only need one output in the FC that connects to the tilt servo. As inputs, we could use one of the AUX channels already available in Cleanflight. The two basic modes we think that would be good to start with are those described above: 1. Dynamic tilting and 2. Fixed arms. However, in the future one might develop other modes or even two axis tilting systems.

The Dynamic tilting mode is the one providing the advantages in aerodynamica and camera view: our vision is to be able to mix the ELEV, throttle (THR) and servo PWMs with rates, expos and such to get different flight behaviours. In some circumstances, however, it might be advantageous to uncouple the servo and elevator PWM 'sign', especially around the neutral (PWM=1500 us), i.e. if releasing the sticks from a fast forward speed might be good to allow the servo to tilt the arms backwards further than the neutral position to improve breaking. This is just theoretical right now and should be tested. Not priority for the first software version in our opinion but would like to experiment it.

We are also thinking of making an algorithm for dynamic tilting to find the best response to the whole system. Using the algorithm or not is optional for the user.

Any other things we might be missing? What do you all think?

Best regards,

        Susheel and Pau

digitalentity, 2015-03-31:
Is this feature available on mass market? I don't think that implementing such a new feature in cleanflight is worthwhile as it may (and probably will) introduce new bugs without giving much of a benefit to an average pilot. I believe we should concentrate on eliminating bugs and supporting widely available equipment before adding support for exotic technologies.

Diastrus, 2015-04-02:
Several similar projects have been already tested with positive results and, as Mech and Aerospace engineer I'm convinced the multirotor scene can still greatly improve in performance if aerodynamics and flight dynamics evolve. However, while many can develop a tilt mechanism, the programming involved is making many private innovators to continue developing it.

We've also investigated the market: after weeks sharing our development in social media networks we've seen an immense interest on bringing this type of system forward in a more user-friendly way. Based on our experience and feedback provided from many multirotor users, we expect the implementation of a servo tilt control routine on Cleanflight to stimulate more people to develop their own tilt system, that's one of our goals. So far we've accomplished to implement a tilt control system in Arduino using a separate Nano board and, although we want to keep it open source for anyone to experiment and improve the code, it would be excellent to get your help guys to bring this in a more user-friendly way in CF as not everyone knows how to code in Arduino.

Just repeating some of the advantages of a tilt system, here our last test this week:

https://youtu.be/-c7OQ8l26QQ

The acceleration is obviously greater due to less mass to move (less innertia) and, to some extend, less aerodynamic drag (canopy being developed). Breaking is equally agressive (will post examples in posterior videos).

My colleague, Susheel, has been already coding the system and we have a basic working code but we hope we can work together and bring this to CF. Thanks.

thenickdude, 2015-04-02:
Why not code it up in a CF fork and get people to use the fork? If the hardware becomes popular then it can be merged back upstream to here. (I guess various interesting tilting motor mounts would be needed)

Diastrus, 2015-04-05:
It sounds good to us. Any link/post you recommend to check to speed up the process?

MauroMombelli, 2015-04-21:
Hi,
i'm coding this system and i need some help and opinion about the work.
Right now we added some different configuration to test some idea, one of this is to keep the body to a fixed PITCH (0° right now but a limit in rotation is planned).
Setting axisPID[PITCH] = 0; in mixer code break the stabilization algorithm, what do you suggest?
I was thinking about saving somewhere the PITCH channel and them set it to 0 BEFORE the PID calculation; but this not seems a clean way to do it, and maybe won't even work.

Diastrus, 2015-04-21:
WARNING: ALL CODE IS UNTESTED

we have officially forked the code, as solution to the problem above you i had to modify mw.c, line 725 to 736: https://github.com/inkonova/cleanflight/blob/tilting/src/main/mw.c#L725 an error that i was making here is "rcData[PITCH] = 1500" ofter setting "rcCommand[PITCH] = 0"; this was giving great problem but I can't find why.

the rest of the configuration stay in the mixer.c (and mixer.h), the servo tilting part is at https://github.com/inkonova/cleanflight/blob/tilting/src/main/flight/mixer.c#L482 and the logic sits in a function "mixTilting" at https://github.com/inkonova/cleanflight/blob/tilting/src/main/flight/mixer.c#L581 called at https://github.com/inkonova/cleanflight/blob/tilting/src/main/flight/mixer.c#L626 ONLY if a tilting configuration has been activated.

Changes in the configurator are very easy, just add the 5 new testing board, but i'm gonna upload it right now.

Code is easy and fragmented in all its component for testing.

MIXER_QUADX_TILT is the base code, it just add a servo contrlled by AUX1 (if AUX1 > 1500ms) or PITCH
MIXER_QUADX_TILT_COS is base + try to add a compensation to roll/yaw because of the rotation of the engine.
MIXER_QUADX_TILT_PITCH is base + try to keep the frame parallel to earth
MIXER_QUADX_TILT_THRUST is base + try to compensate for the loss of trust because of engine rotation
MIXER_QUADX_TILT_ALL is all of the above and candidate to be the (only?) tilting configuration

I need also some clarification for the next step; it is a bit unclear how can I add a filed to save on the configurator, it seems like all config is send (and probably received) with a fixed shared structure, any advice is appreciated.

thenickdude, 2015-04-22:
@Diastrus it looks like you've made a lot of whitespace changes on lines that don't need to be modified:

https://github.com/cleanflight/cleanflight/compare/master...inkonova:tilting

Cleanflight code standards require 4 spaces as an indent rather than tabs, and most of these lines should not have been touched (it makes it very hard to follow diffs otherwise).

`rcData` is the raw readings from the rc sticks before RC Rates and stick expo curves are applied, which becomes `rcCommand`.

For adding new fields to the configuration, the easiest way is to get users to set them by the CLI, and add your field to the structures in serial_cli.c so that it becomes available to modify by `set` commands.

Diastrus, 2015-04-22:
@thenickdude 

Thanks for the feedback, we are going to be more careful with the syntax. We'll come back soon with details about what we intend to do.

Ping @MauroMombelli @susheelpuranik

MauroMombelli, 2015-04-22:
@thenickdude thanks for your time.
Probably eclipse's code auto-formatting, can you share your formatter's configuration?

i get what ~~rcData~~ and ~~rcCommand~~ does, but if you try to force one of the field of 'rcData' to 1500 weird things happen

We are still looking for the field to set up, but basically we found out our servo PWM range is from 500us to 2500us for almost full 180°, and then we want the user to limit the angle, 60/70° was good for @Diastrus during testing.

Also when jumping from dynamic to fixed tilted servo (it can be done while flying), the expo curve applied on pitch should change; any advice as best practice?

ah, we are using the Servo 0 in the AfroFlight32 but that we the user loose the gimbal, is there a way to add a special case like for gimbal and use different pin? Setting one as ~~MAP_TO_SERVO_OUTPUT~~ in the pwm_mapping does not seems do have good results, probably i have to take into account some HW/timers limitation, is this correct? has anyone a table of what configuration are permitted?
 
Also why (for the Naze) ~~multiPPM~~ is all defined as ~~MAP_TO_MOTOR_OUTPUT~~ but then PWM 1 and 9 are overrided as ~~MAP_TO_SERVO_OUTPUT~~? I mean, why just not create a separate array with ~~#ifdef~~? see https://github.com/inkonova/cleanflight/blob/tilting/src/main/drivers/pwm_mapping.c#L78 and
https://github.com/inkonova/cleanflight/blob/tilting/src/main/drivers/pwm_mapping.c#L427

t0mmg, 2015-04-26:
Just to add, here's a link to another DMT frame in the pipeline: http://onlyflyingmachines.com/product/ofmhyper330/

Diastrus, 2015-04-26:
@t0mmg Thanks for the heads up.

I guess this is just justifying a bit more the inclusion of a tilt control on Cleanflight. According to our research for the last 12 months, we actually expected that and more products taking off during spring/summer 2015.

MauroMombelli, 2015-04-27:
@thenickdude for the tilting motor we want to add a limit to the rotation, and that can be different form positive and negative. i've added the 2 byte (uint8_t as it is in degree, 0-180) to the structure, the added the read and the print in the serial_cli.c and works fine, then when adding to to serial_msp.c i can read and display data on CLI, but when i try to write the code blocks.

meanwhile found the bug at #172

MauroMombelli, 2015-04-27:
i have some other question for you.
The board we are using is a AfroFlight (Naze) with input PPM.
Actually we are having some problem with the pin to use.
We use the camera's gimbal so it should work on X4, X6 and X8; but you lose gimbal.
as far as i can understand 
on the x8 does NOT have any other free pin, we have to loose the gimbal;
on the x4 there is a full 4 pin witch can be used as Servo;
on the x6 we can use a timer witch command 4 pin and move the gimbal there, with our tilting, and leave the now free 2 pin (TIM1) for the engines.

We would like to have your feedback on what you think to be the best way to implement that

michel-pet, 2015-04-28:
[image]
[image]
[image]

i hope we can test the new software soon.

l2smith2, 2015-05-13:
Why not make this work as a VTOL system for airplane mode. Obviously set motor tilt limits for quad mode but allow a full forward tilt in airplane transitioning yaw to throttling motors on the left or right. Also allowing the shut off of select motors once past a certain tilt point if need.

There's no need for a servo gimbal if the frame remains level

MauroMombelli, 2015-05-13:
@l2smith2 : thanks for the input, we considered it but the frame does not have any real lift surface, so the plane logic won't kick in.
if you add some lifting surface you get something very different machine, with its good and bad.

Meanwhile you can see that today i updated the fork of cleanflight and configuration with a new BIG change, i've removed varius configuration for testing but one, and this one has all the feature that can be enabled and tweaked with a new tab in the configuration.

Again this break the compatibility with cleanflight configuration, when flashing this software you have to fully erase the flash; does not seems to be any check on cleanflight and/or cleanflight-configurator that the configuration's saved  version match the one used by the firmware; I'll take a look and fix this in a different pull request, if this is the case.

@thenickdude  @hydra  if everything is fine we plan to make a pull request in two week, I just aligned the code (branch tilting on my side!) with current master/development of the cleanflight if you want to have a look.
The only part that i'm concerned may touch you is the new way to use the Servo; not only you have to set the endpoint (-90°, 0° and 180°), but then you can specify a "software limit" to desired angle. This is needed because to apply some compensation we need to estimate the angle of the servo; this can also be really helpful for example on "camera gimbal" to follow correctly the frame orientation, for example, and limit the excursion to prevent hardware limitation. maybe I can prepare a pull request with this edit and send it as soon as possible, so you can start to evaluate/merge it; meanwhile i can continue working on the tilting code.

CAUTION IF YOU WANT TO TEST; especially that last commit (and is a big one) HAS NOT BEEN TESTED: if everything is good we hope to release the pull request for the end of the week

Btw we release a teaser video and started the pre-sell, here the official page: http://www.tiltdrone.com/

l2smith2, 2015-05-13:
Sounds good. I just thought it'd be good if the VTOL option was there for
people who wish to build more of a plane with quad motors than a tilting
motor quad.

Since no other firmwares support vtol it'd be great to see cleanflight be
the first.

michel-pet, 2015-05-14:
https://www.youtube.com/watch?v=r5iw2UdID6M&feature=share
test setup have to set the servo 100% ok and look at the setings.

MauroMombelli, 2015-05-26:
please see pull request #872

MauroMombelli, 2015-11-07:
@hydra @thenickdude after some testing, we found oout that even if input pitch is forced to be 0, so the body should always be horizontal, we are experiencing body tilt during acceleration/deceleration.

I was thinking it may be the acceleration "breaking" the orientation algorithm, i have some black box data to share, and the idea was to try to use the DCM and see at different output result for orientation starting from raw data.

Do you have any idea of what else can create that issue?

Anyway porting to 1.10 is done, i'm just waiting to run some more test on the field before sending the new pull request.

joshuabardwell, 2015-11-07:
Pardon me if this is obvious, but the body tilt during acceleration/deceleration may be related to a known issue where the copter is in autolevel mode and its pitch is disturbed when decelerating rapidly. The accelerometer can't distinguish between gravitational acceleration and other accelerations. So when you accelerate or decelerate, the copter may pitch forward or backward in response to the additional forces generated. Likewise in a turn--if you go through a coordinated turn, the G vector will point straight down through the copter's Z axis. This makes the copter think it's level, so it will want to roll into the turn. You can also see this if you have an artificial horizon on your OSD--it will tend to roll out of a turn as it thinks the copter is more level than it is due to the acceleration of the turn.

I fly pretty much exclusively in rate mode, so I haven't experienced these myself, but I've heard others describe them, and conceptually, it makes sense.

MauroMombelli, 2015-11-07:
this is what i mean by 'acceleration "breaking" the orientation algorithm'; the sensor fusion should take care of those thing, especially because out quad can reach crazy acceleration.
Then the solution to try to use DCM to see if i get better result is a good idea. Anoher truick can be exclude the accelerometer reading when they are too different than 1G; tit seems this problem is visible when we get acceleration in range of 1G, so let's say accept accelerometer reading in range 0.8 < X < 1.2?

@digitalentity can you give me a stable commit with DCM to port the tilt code and test that in real life?

digitalentity, 2015-11-08:
@MauroMombelli please have a look at #1411. It's been flight-tested in betaflight and iNav and verified to be working better than current complementary filter implementation.

thenickdude, 2015-11-09:
Presently the accelerometer is ignored when the reading lies outside 0.85 <
x < 1.15 g.

MauroMombelli, 2015-12-09:
Sorry for late update, many thing has going on, right now porting on the 1.10 is going on testing, and as training course @AlexGratner develop a new feature that we call "multi modal stability" or MMS, in witch the quad can hover keeping the body to a desired angle (driven by a pot). Still based on 1.8.6 that you can find, while we test and polish it, on https://github.com/inkonova/cleanflight/tree/tilting-1.8.1

here the maiden fly: http://youtu.be/3R1DIH9kmCQ

flegmatoid, 2016-08-01:
Is there any update on this?
Found similar implementation on APM, they name it [ptilt mode](https://github.com/mkschreder/apm-px4-tilt/commit/e874d5cd72beafda52027de40a0973812f9f7a3f)

fafa80, 2016-08-09:
Hi, will the tilt function ever be implemented in cleaflight 1.13.0?
Which firmware may I try in order to setup a tilt rotor?
I have a CC3D and a SP F3.
Thanks

MauroMombelli, 2016-08-12:
the working repository is https://github.com/inkonova/cleanflight, but
there is only for Naze32; compiling is not a big issue if you have a linux
machine, but unfortunately testing is another pair of shoe.

closed 2017-02-25 by hydra