cleanflight/cleanflight #693: Telemetry Stopped Working with 1.8

TricopterFan, 2015-03-31:
I just upgraded Cleanflight from 1.7.2, to 1.8 and my telemetry stopped working, more accurately, the aircraft battery voltage is not being sent to my transmitter. I'm using a d4rii and a taranis x9d with opentx. I'm not using any serial port stuff, I just want to see my voltage on my transmitter. I made sure feature telemetry was enabled, and vbat. The naze32 can see the voltage still, but not my transmitter. It worked fine before the upgrade.

thenickdude, 2015-03-31:
On the Ports tab, do you have FrSky selected in the telemetry column on UART1? (assuming that you're using the FrSky pins)

TricopterFan, 2015-03-31:
No I didn't, that fixed the issue. Thanks!

closed 2015-03-31 by TricopterFan