cleanflight/cleanflight #695: Improvement for stock PID mode 3 Horizon code

ctzsnooze, 2015-03-31:
Hi - and thanks for Cleanflight!  :-)

I have written and tested a significant improvement to the classical MultiWii 2.3 Horizon mode code.  It improves controllability at high quad angles and generally improves acro functionality of Horizon mode.

It simply involves replacing the stock one line of code with this three-line variant.  The intent is to reduce the self-levelling as the quad becomes closer to vertical orientation.

Original code:

    if (FLIGHT_MODE(HORIZON_MODE)) prop = MIN(MAX(ABS(rcCommand[PITCH]), ABS(rcCommand[ROLL])), 512);

Replace with this:

    if (FLIGHT_MODE(HORIZON_MODE)) {
        prop = MIN(MAX(ABS(rcCommand[PITCH]), ABS(rcCommand[ROLL])), 512);
        prop = prop + (MIN(MAX(ABS(inclination.raw[PITCH]),ABS(inclination.raw[ROLL])),1800)>>3);
        prop = MIN(prop,512);
    }

The original code generates a 'prop' value up to 512 from the angle of the sticks.  'prop' is used to reduce self-levelling, so that when 'prop' is 512 - in the normal code, at full sticks - there is no self-levelling at all. 

The problem with this is that to get (say) steeply angled forward flight, stick angles must be a long way forward, and then the quad becomes dynamically unstable - the more forward the stick gets, the more the quad starts rotating forward even if the sticks are held constant (because levelling is reduced), so then you must ease the stick back a bit... but then not only does the desired angle decrease, also the levelling comes back, and it starts to come back more than intended... but if you add more stick forward again, it pitches too far forward.  The end result makes it very difficult to keep steeply angled flight stable. 

The proposed new code adds another proportional reduction to the levelling based on the actual angle of the quad.  For example, 112 is added when vertical and 256 when inverted (approx).   

The 'flyability' of the quad in Horizon mode with this change becomes much nicer, especially at steep angles or when performing loops or flips.   It is possible to maintain an angle of (say) 80 degrees from horizontal and control that angle nicely with the stick.  This is not at all easy with the stock code.

Anyway this only works with the multiwii23 code.  I have several other people test it and they say it's great.  Feel free to test it out and see if you agree.

I would really love to see it in Cleanflight maybe as a CLI option?

If that is at all possible I'd be very grateful.

Thanks

Chris

hydra, 2015-03-31:
thanks for the detailed description of the problem and your solution. we need some testers of this.  @nebbian @thenickdude ?

ctzsnooze, 2015-04-01:
The only problem is that almost no-one any good flies Horizon very much!   :-)

Some experienced testers would be great, every single person to date loves it.  I enabled a Tx switch in MulitWii to swap between the classic and new code, until I got it how it seemed best.  I didn't know how to do that on my cleanflight build... if you could let me know how to do an 'if AUX2 >1200 then new code else classic code' statement, that could help with testing.  :-)

In multiwii 2.3 I found that the inclination values were 10 per degree.  If instead of dividing by 8, I also tried dividing by 4 (i.e. >>2)... then you can just about fully acro when fully inverted, with no levelling, when vertical you will be fully acro at half stick.  This way you only get levelling when both sticks and quad are close to level.  Kind of a weird feeling and didn't work so well like that.  Testers may find some variant that works better.

When testing there is no difference at small angles.  Only when doing big power circles at high angles, or fast forward punch-outs at steep angles, does it become apparent that stock Horizon just doesn't work all that nicely at all, and this modification helps with that immensely.  I hope the testers agree.

Cheers and thank you again!

thenickdude, 2015-04-01:
@ctzsnooze this is a great idea, I definitely experienced the "tilt more.. a little more... *sudden roll*" when I was flying Horizon.

ctzsnooze, 2015-04-01:
Please give the code above a try and post your feedback here, that would be excellent.  Yes it was scary if it would not level at all while inverted!  :-)

ctzsnooze, 2015-04-02:
My blog entry with some videos of what its like to fly the new code:

http://www.rcgroups.com/forums/showthread.php?t=2383085

nebbian, 2015-04-02:
To be honest I wouldn't notice much difference.  I fly with insane stick scaling anyway, so all the way up to 90 degrees attitude is easily controllable, because my stick is only at about 30% travel at that point.  This means that the self level code is still quite active even when fully banked over.  I'll try this code over the weekend and see what difference it makes.  I am a bit concerned that it won't self level after a fast series of multiple flips (I like to do several from a decent height and centre the stick to level out).

Jahnkeanater, 2015-04-02:
I have noticed that flips don't work the same as they did in multiwii. When rolling the quad will get stuck inverted for a little bit before it finally decides to finish the flip. corner flips work well but the quad tends to end up in a different heading at the end. possibly caused by the fact that a dx6i can not reach the endpoints 1000-2000 even at 125% throws. I can only get 1070-1930.

ctzsnooze, 2015-04-07:
This happened to some extent in multiwii - my horizon code seems to fix it, though not completely

Jahnkeanater, 2015-04-11:
I fly horizon mode all the time. is this fix implemented in v1.8.1? I would be willing to test it on my quad.

Jahnkeanater, 2015-04-11:
I have also noticed that when I forget I'm in angle mode and I try to do a flip my quad does something at a very high speed and ends up facing the exact opposite direction. I would explain it in more detail but it happens so fast I really cant tell.

hydra, 2015-05-15:
@Jahnkeanater blackbox logs and video please.

hydra, 2015-05-15:
Can some people try the fix (above) and give feedback?

ctzsnooze, 2015-05-19:
I've now written similar code for LuxFloat that works even better. Just amazing.   Etheli has confirmed it works well.  See this (not very good) video where I demonstrate how it is now possible to do basic inverted stuff, slow loops, big arcs etc: https://www.youtube.com/watch?v=weGS6ljycVk

With floating point maths it is possible to fine-tune the angle at which all levelling stops, something I can't do in the MultiWii code with its binary divisions.

The code from flight/pid.c is pasted below, including the original code commented out.

Note that I am using a value of '3.5' to set the angle at which all levelling stops.  The level of the quad is reported as 1800 when inverted, dividing that by 3.5 gives 514.  This reduces the horizon_level_strength factor down from 500 (normal) to 'below zero' when the quad is vertical, becoming zero from within 5% of vertical.  So you can fly inverted and do big slow loops etc.  

	// Progressively turn off the horizon self level strength as the stick is banged over
	// ORIGINAL CODE
	// horizonLevelStrength = (float)(500 - mostDeflectedPos) / 500;  // 1 at centre stick, 0 = max stick deflection  ORIGINAL
	// END ORIGINAL CODE

	// MODIFIED HORIZON CODE CTZSNOOZE
        horizonLevelStrength = (float)(500 - mostDeflectedPos);  // 500 at centre stick, 0 = max stick deflection
        horizonLevelStrength = horizonLevelStrength - (float)(MIN(MAX(ABS(inclination.raw[PITCH]),ABS(inclination.raw[ROLL])),1800))/3.5f;  
		horizonLevelStrength = constrainf ( horizonLevelStrength, 0, 500);  // limit to range 0 to 500
		horizonLevelStrength = horizonLevelStrength / 500;  // 1 for full levelling and 0 for none
		
	// reduces levelling by sticks as usual but also according to inclination of quad
	// angle calculation returns about 900 when vertical 1800 when inverted
	// divide 1800 by 6 removes 150 when vertical and 300 when inverted
	// divide by 3.5 as above removes 257 when vertical and 514 when inverted - so full acro for 5 or so degrees around fully inverted

I am sure better coders could make neater code!  :-)

It would be really great if the angle at which all levelling would stop could be configured in the CLI.  So it would be nice, in the above example, if I could just set '175' in the CLI and have no levelling as soon as the quad went past 175 degrees.  Or, I could set say 160 and have an even bigger inverted 'no-levelling' window, or say 200 and retain some levelling at 180.  This way the user could tune the inhibition of levelling as desired.  And if the value was '0' then the entire code would not run, be ignored, and the standard code would be active.  

That's how I'd hope it could be implemented, I just don't know how to do it best.

This code could easily be ported to all Horizon code using floating point math.

For PIDC3 the previous code can be used but would need to be written differently to take advantage of a CLI setting value.  I'm sure it could be done I don't know how.

Sorry that I am not more expert at programming.

ethomas997, 2015-05-19:
Yeah, the flight improvement is substantial; I like it lot.  My plan is to work on turning this into a pull request that implements the functionality across multiple PID controllers.

--ET(heli)

ctzsnooze, 2015-05-19:
Mate that would be absolutely fan-bloody-tastic!  :-)

ethomas997, 2015-05-25:
Here is my initial PR for this:  #948 
--ET

closed 2017-02-25 by hydra