cleanflight/cleanflight #696: More than one VBat Scale Factor - maybe stored in the PID profile somehow?

ctzsnooze, 2015-03-31:
Often people use different size batteries on one quad.  For example, 2S for docile flying, 3S for faster; or for Acro quads, 3S for a bit of fun, 4S for manic high speed sessions!  :-)

At the same time, its really good to know the battery voltage.  The internal VBat monitoring code is really great for this.  If a large external LED is added, the flashing of the LED can let you know its time to land.  Or you can drive a buzzer very easily.  Many 32 bit boards include the hardware voltage dividers required to provide an input voltage for the VBat pin.  I use VBat alarms with external LEDs a great deal, especially on ultralight quads where I want to use a simple sat Rx, and they are invaluable.

Anyhow, the problem at the moment is that one has to specify one, and one only, value for the VBat scaling parameter.  Because the voltage divider is set once only in hardware, and there is only one scale factor, the user may set the VBat scale value to work properly with 3S - but then it will be no use at all for 4S, or 2S.  

One possible solution is to provide a means to store, optionally, an alternative VBat scale value in each of the 3 PID profiles.  If set to a non-zero value in the profile, the value in the profile replaces the default VBat scale value.  

This way the user can create a '2S' profile with a VBat scale value  that works with 2S, and another '3S' profile with the correct VBat scale for 3S.  

This would be really great, because the LED or Buzzer based alarms could then work really well with any sized battery.  Anyway with different sized batteries, slightly different PID values may be needed, so it makes sense to also fix the battery monitoring.

I'd be incredibly grateful if this idea one day found the light of day!  :-)

Thanks for considering it!

Chris

ledvinap, 2015-03-31:
vbat_scale should be set to match voltage divider, so CF calculates correct voltage.
All thresholds are calculated on per-cell basis, so it should work.

Small problem is that cell detection assumes vbatmaxcellvoltage (defaults to 4.3v).  When starting with discharged high cell pack, the cell count detected may be wrong ...

hydra, 2015-03-31:
The only thing that might be useful is per profile battery capacity for telemetry fuel readouts not the vbat scale.

The FC could be made to beep out how many cells it thinks it has though.

hydra, 2015-03-31:
Perhaps having battery cell count in the profile might be used. And if unset it can auto-detect as it does now.

ctzsnooze, 2015-03-31:
Oh I didn't realise it performed auto-detection of number of cells.  I had assumed - bad thing to assume - that the device simply got the scale factor then assumed a fixed number of cells (in my case it seemed to work well if the displayed voltage was 4S when the real voltage was 2S... now I know why!)  

Auto-detection is cool, but I agree it could get it wrong if starting with a flat battery.  Therefore, yes, being able to set the cell count in the profile could be quite useful.  Apologies for not properly understanding how it works!  :-)

ledvinap, 2015-03-31:
@ctzsnooze : 4S battery must be discharged to < 3.25V/cell to be detected as fully charged 3S, so it should be minimal problem on 4S. On 6S it is < 3.6V/cell, still acceptable (it's no-load/light load voltage).

ctzsnooze, 2015-04-01:
Yep that sounds good.  Honestly I didn't realise how clever the automatic cell voltage code is.  Sounds like this isn't an issue at all.  Thanks for reading and considering the idea - I apologise for troubling you.

Maybe the battery.md info file could explain it a little differently to help people like me out?

At present it says:

Minimum and maximum cell voltages can be set, and these voltages are used to detect the amount of cells you are using.

maybe the following could be a subtle and slightly clearer alternative:

Minimum and maximum cell voltages can be set, and these voltages are used to auto-detect the number of cells in the battery when it is first connected.

Thanks again, you guys are great!  :-)

commit dee68f5e (2015-05-15): Clarify Battery voltage detection statement in docs.  Fixes #696.
closed 2015-05-15 by hydra