cleanflight/cleanflight #700: Documentation "Building in Mac OS X"

Petfrogg, 2015-03-31:
In the end of https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Mac%20OS%20X.md it says:

make clean TARGET=NAZE
make TARGET=NAZE

I suggest change it to the following:

In case of NAZE
make clean TARGET=NAZE
make TARGET=NAZE

For other available boards type:
make help

CC3D and Openflight as flashing tool
If you are using a CC3D board and updating the firmware using Openflight you need to compile it using these lines in order to get it to work:
make TARGET=CC3D OPBL=yes

thenickdude, 2015-03-31:
I don't think these instructions belong in the Mac-specific building notes, we probably need a more general doc file for them.

hydra, 2015-04-01:
@thenickdude agreed.

@Petfrogg can you perhaps submit a pull request and create a general 'Building.md' file with this information and add links from it to the platform specific sections/files?

thenickdude, 2015-05-07:
This could probably fit in https://github.com/cleanflight/cleanflight/blob/master/docs/development/Development.md now

closed 2017-02-25 by hydra