cleanflight/cleanflight #701: Setting MSP on SOFTSERIAL1 when LED_STRIP is enabled causes infinite loop

thenickdude, 2015-04-01:
On the Naze32 running 1.8.0 final, I loaded this perfectly reasonable configuration:

```
# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature VBAT
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature MOTOR_STOP
feature LED_STRIP
feature ONESHOT125
feature BLACKBOX

# map
map TAER1234

# led
led 0 11,0::IWC:2
led 1 10,15:S:WC:2
led 2 9,15:S:FT:0
led 3 8,15:S:AT:2
led 4 7,15:S:AT:2
led 5 6,15:S:FT:0
led 6 5,15:S:WC:2
led 7 4,0::IWC:2
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 3500
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set input_filtering_mode = 0
set min_throttle = 1100
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 1
set auto_disarm_delay = 5
set small_angle = 25
set flaps_speed = 0
set fixedwing_althold_dir = 1
set serial_port_1_functions = 1
set serial_port_1_msp_baudrate = 5
set serial_port_1_telemetry_baudrate = 0
set serial_port_1_blackbox_baudrate = 5
set serial_port_1_gps_baudrate = 4
set serial_port_2_functions = 64
set serial_port_2_msp_baudrate = 5
set serial_port_2_telemetry_baudrate = 0
set serial_port_2_blackbox_baudrate = 5
set serial_port_2_gps_baudrate = 4
set serial_port_3_functions = 0
set serial_port_3_msp_baudrate = 5
set serial_port_3_telemetry_baudrate = 0
set serial_port_3_blackbox_baudrate = 5
set serial_port_3_gps_baudrate = 4
set serial_port_4_functions = 0
set serial_port_4_msp_baudrate = 5
set serial_port_4_telemetry_baudrate = 0
set serial_port_4_blackbox_baudrate = 5
set serial_port_4_gps_baudrate = 4
set reboot_character = 82
set gps_provider = 0
set gps_sbas_mode = 0
set gps_auto_config = 1
set gps_auto_baud = 0
set serialrx_provider = 1
set spektrum_sat_bind = 0
set telemetry_switch = 0
set telemetry_inversion = 0
set frsky_default_lattitude = 0.000
set frsky_default_longitude = 0.000
set frsky_coordinates_format = 0
set frsky_unit = 0
set frsky_vfas_precision = 0
set battery_capacity = 0
set vbat_scale = 112
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 34
set vbat_warning_cell_voltage = 34
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = 0
set current_meter_type = 1
set align_gyro = 0
set align_acc = 0
set align_mag = 0
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 90
set max_angle_inclination = 500
set gyro_lpf = 42
set moron_threshold = 32
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set yaw_control_direction = 1
set pid_at_min_throttle = 1
set acc_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = 1

# dump profile

# profile
profile 0

# aux
aux 0 0 2 1500 2100
aux 1 1 0 1700 2100
aux 2 2 0 1300 1700
aux 3 12 1 900 1500
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900
aux 20 0 0 900 900
aux 21 0 0 900 900
aux 22 0 0 900 900
aux 23 0 0 900 900
aux 24 0 0 900 900
aux 25 0 0 900 900
aux 26 0 0 900 900
aux 27 0 0 900 900
aux 28 0 0 900 900
aux 29 0 0 900 900
aux 30 0 0 900 900
aux 31 0 0 900 900
aux 32 0 0 900 900
aux 33 0 0 900 900
aux 34 0 0 900 900
aux 35 0 0 900 900
aux 36 0 0 900 900
aux 37 0 0 900 900
aux 38 0 0 900 900
aux 39 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# servo
servo 0 1020 2000 1500 30 -1
servo 1 1020 2000 1500 30 -1
servo 2 1020 2000 1500 100 -1
servo 3 1020 2000 1500 100 -1
servo 4 1020 2000 1500 100 -1
servo 5 1020 2000 1500 100 -1
servo 6 1020 2000 1500 100 -1
servo 7 1020 2000 1500 100 -1

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = 1
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = 1
set deadband = 6
set yaw_deadband = 6
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_direction = 1
set tri_unarmed_servo = 1
set servo_lowpass_freq = 400
set servo_lowpass_enable = 0
set default_rate_profile = 0
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1200
set failsafe_min_usec = 985
set failsafe_max_usec = 2115
set gimbal_flags = 1
set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff = 5.000
set acc_unarmedcal = 1
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf = 0.600
set baro_cf_vel = 0.985
set baro_cf_alt = 0.965
set mag_declination = 0
set pid_controller = 3
set p_pitch = 32
set i_pitch = 30
set d_pitch = 23
set p_roll = 32
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 50
set d_yaw = 0
set p_pitchf = 2.500
set i_pitchf = 0.600
set d_pitchf = 0.060
set p_rollf = 2.500
set i_rollf = 0.600
set d_rollf = 0.060
set p_yawf = 8.000
set i_yawf = 0.500
set d_yawf = 0.050
set level_horizon = 3.000
set level_angle = 5.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 30
set tpa_rate = 25
set tpa_breakpoint = 1500
```

This configuration has LED_STRIP enabled (and SOFTSERIAL disabled). In the configurator's ports tab, if I now tick "MSP" on SOFTSERIAL1 and save (obviously not allowed because SOFTSERIAL isn't enabled), the board hangs on boot and cannot be connected to. It goes into an infinite loop with this stack trace:

```
#0  timCCxHandler.6801 (tim=0x40000400,
    timerConfig=0x20002e18 <timerConfig+80>) at ./src/main/drivers/timer.c:733
#1  <signal handler called>
#2  0x08001de8 in TIM_OC1Init (TIM_OCInitStruct=<synthetic pointer>,
    TIMx=0x40000400)
    at ./lib/main/STM32F10x_StdPeriph_Driver/src/stm32f10x_tim.c:289
#3  ws2811LedStripHardwareInit ()
    at ./src/main/drivers/light_ws2811strip_stm32f10x.c:71
#4  ws2811LedStripInit () at ./src/main/drivers/light_ws2811strip.c:82
#5  ledStripEnable () at ./src/main/io/ledstrip.c:1044
#6  init () at ./src/main/main.c:378
#7  0x0801274a in main () at ./src/main/main.c:468
```

hydra, 2015-04-01:
Thanks for the report.  Are you looking into it or shall I look?

thenickdude, 2015-04-01:
I'm not really clear on the whole timer setup code, so you might have better luck than me.

hydra, 2015-04-01:
Ok, maybe @ledvinap has an idea of what might be happening here?

ledvinap, 2015-04-01:
Seems that softserial got registered as overflow handler. If timer3 is configured to 800kHz in ws2811LedStripHardwareInit(), it will fire faster than softserial can handle it ...

ledvinap, 2015-04-01:
@thenickdude : Yes, findNextSerialPortConfig() is probably the cause ...

PR #703 (2015-04-01): Don't allow softserial ports to be opened if softserial isn't enabled
Closes #701 

(MSP still tries to open the softserial port it is erroneously configured for, but now the open harmlessly fails and it doesn't try to use the port further)
closed 2015-04-01 by hydra