cleanflight/cleanflight #702: HoTT Telemetry working but voltage warning not

usab, 2015-04-01:
Not sure if this is the right place for this issue.

I successfully set up a HoTT receiver with latest 1.8.0 cleanflight firmware and telemetry connection to transmitter.
The overall voltage is displayed (11.9v). The buzzer (directly connected to naze32) is peeping fine.

The only problem is that the transmitter does not beep. (that's THE advantage of having telemetry)

Are there known problems? VBAT is enabled (and the naze32 buzzer is working, so the settings should be fine).

Thanks for your help!

hydra, 2015-04-01:
@Pierre-A Any chance you could take a look?

Pierre-A, 2015-04-01:
yes sure, this is the object of PR≠539..

hydra, 2015-04-01:
@Pierre-A Ahh yes, i thought i remembered seeing something.

Now that the serial port re-work is done I should have time to look more into the smartport and hott telemetry systems and related PRs soon.

Pierre-A, 2015-04-01:
OK,
just waiting for that to open a new PR for hott_TextMode

hydra, 2015-04-01:
@Pierre-A I've merged #539 now.

hydra, 2015-04-01:
@usab Please test v1.8.1 firmware when released and re-open if there's an issue.

closed 2015-04-01 by hydra

usab, 2015-04-01:
Thank you both so much for this awesome software. :)

I will test the new code as soon as possible!

Pierre-A, 2015-04-02:
**@hydra**  looking at hott.c in the v1.8.1, the PR≠539 is not included (and so has not been merged).. something wrong somewhere ?

usab, 2015-04-02:
I just installed 1.8.1 and tried to get it working. The transmitter still doensn't beep. :(
There is also no information in the changelog about the fixed HoTT telemetry.

Is there a problem with the code? (As Pierre-A said)

usab, 2015-04-05:
@Pierre-A 
I have a short question. How do I change the interval of the warning sound when flying with HoTT telemetry? Is it possible to set it to "1s" (beep every one second) instead of i.e. 5s?

Pierre-A, 2015-04-05:
**@usab** this is included in PR≠539, I introduced the parameter *Hott_alarm_sound_period* you can change as you like in Configurator CLI (from 0  to 120 s). You'll get it when merged.

hydra, 2015-04-07:
@Pierre-A i think perhaps I merged it and didn't push it to the repo.  Let me sort this out now.

usab, 2015-04-07:
Thanks hydra for checking this! :)

hydra, 2015-04-07:
Merged, please test and re-open if there's still an issue.

hydra, 2015-04-07:
@Pierre-A Please see 7d9fc2699a864fed09416298c581e6480bc7114b and 0456444cfbec05913345b87f263290330f4c14f1

Pierre-A, 2015-04-08:
thks for merging; 
the alarm sound works fine for me (Naze32 and GR16 + Graupner MX20) with the proper time interval as defined in CLI.

usab, 2015-04-08:
How did you do this? Is there another version then 1.8.1 available?

Pierre-A, 2015-04-08:
As the 1.9.0 version is not yet available, I compiled a .hex from the latest commit of the source code (87091802e6).
PM me if you want the file, I'll send it to you with pleasure.

usab, 2015-04-08:
Thank you, Pierre!

Seems to work fine! :+1:

Pierre-A, 2015-04-08:
you're welcome !