cleanflight/cleanflight #705: looptime set to 0

handsomejackuk, 2015-04-01:
Any reason why looptime is set to zero after flashing version 1.8 of cleanflight to naze 32 board.. i have put it back to 2500 just wondered why ?

thenickdude, 2015-04-01:
How did you transfer your config over to the new firmware (or did you not configure it at all after flashing?)

hydra, 2015-04-02:
this could be an issue with the configurator restore, if you used backups please post the backup files in a ticket for the configurator.  The default looptime was not changed.  New MSP commands were added to backup/restore looptime.

closed 2015-04-02 by hydra

handsomejackuk, 2015-04-03:
i checked after flashing and was surprised it was zero, as i never run at zero.... i did a restore of the settings, but it was zero after flashing no concern just wanted to make you aware of this if it was an issue....thanks for the continued hard work...

ralenekov, 2015-04-04:
When doing a backup from 0.63.0 configurator using the 1.7.x firmware, it recorded looptime=0 in the backup file. A dump from CLI works fine though. I noticed it when migrating to 1.8.x.