cleanflight/cleanflight #706: CC3D does NOT initialize when using Satellite

catchthewind, 2015-04-01:
I have 1.8 Cleanflight uploaded to CC3D and configured with Orange DSMX Satellite. The problem is that CC3D does not initialize when my Transmitter is ON.  

USB connection also does not work when transmitter is ON and Satellite connects during CC3D initialization.

The only way to arm the CC3D or use it with PC on USB is to turn it on without working transmitter.
Then initialization goes ok and transmitter bind could be done later.

This is not fully safe and also a lot of hassle to turn the radio off between every battery and every PC setup. I use naze with cleanflight on my other copter and this problem does not exist there.

thenickdude, 2015-04-01:
What port is your Rx connected to? Maybe just post your whole "dump" output from the CLI tab.

catchthewind, 2015-04-01:
I have orange satellite connected to LEFT port described FLEXI PORT.

catchthewind, 2015-04-01:
# version
Cleanflight/CC3D 1.8.0 Mar 16 2015 / 23:45:38 (8e65707)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature VBAT
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE
feature ONESHOT125

# map
map TAER1234

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
[... 241 lines cut ...]
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

thenickdude, 2015-04-01:
As a diagnostic, please try disabling the FAILSAFE feature and seeing if the problem still persists.

catchthewind, 2015-04-02:
After disabling failsafe the problem still exists.  
Later on I will try the different satellite cause on naze/flip board I have used lemonrx sat and here I have orangerx on cc3d. I need to make sure if the problem is CC3D related or ORANGERX Satellite related.

thenickdude, 2015-04-02:
Hm I suppose in theory the satellite could be browning out the 3.3V regulator on the CC3D, causing boot to fail. Sorry, I don't have a sat here to test it with, hopefully someone who does can chime in.

catchthewind, 2015-04-02:
I have tested lemonrx satellite as well as original dsmx spektrum sat and problem still exists on CC3D. On flip32 board I am using lemonrx sat without any problems and 3.3V is also taken directly from the board. 

The problem is CC3D related and forcing me to power cc3d first and then radio :(

catchthewind, 2015-04-02:
I also had tail servo problem on CC3D described here https://github.com/cleanflight/cleanflight/issues/586  It looks like CC3D has more problems than naze/flip boards...

hydra, 2015-04-02:
@catchthewind please keep the issue focussed on on topic.  Problem is likely your CC3D board not providing enough power to the RX at boot.  I'll test it with my one and see what happens.

catchthewind, 2015-04-02:
I have tested with the another brand new CC3D board and it acts exactly the same. 
@hydra  Please let me know how your test goes?

catchthewind, 2015-04-10:
@hydra  

1. Have you had a chance to test it yourself? 
2. Is it working ok on your cc3d?
3. Do you think this is something which could be fixed or its a hardware problem and I should replace my cc3d with flips32?

hydra, 2015-04-28:
@catchthewind 

1 & 2 not had a chance to test yet.
3 possibly fixed in software by adding a delay to the startup code.  you could try this yourself, see main.c init()

closed 2017-02-25 by hydra