cleanflight/cleanflight #708: Control endpoints are unatainable

Jahnkeanater, 2015-04-02:
With a DSMX satellite receiver and a DX6i +/-125% throws only gives 1070-1930 on all inputs. 1000-2000 should be easily attainable. sparky with FW v1.7.2.

xorrbit, 2015-04-22:
With a DX5e I was only getting 1099 - 1938 or so. I was able to use my throttle by setting MIN_CHECK to 1150 via the CLI. This sets the value under which the motors will stop spinning (if you have MOTOR_STOP on) and arming/disarming will occur.

digitalentity, 2015-04-22:
Both of you should probably benefit from RC calibration (#618) that adds ability to map any input from RC receiver to 1000-2000 expected by CF firmware.

xorrbit, 2015-04-24:
Yup, that looks like it would solve this issue without having to mess with MIN_CHECK.

closed 2017-02-25 by hydra