cleanflight/cleanflight #709: GPS 3D fix status > LED Strip

skaman82, 2015-04-02:
Is there a way to indicate 3D fix on the LED Strip? (naze32)

hydra, 2015-04-28:
no, but it's a good idea.  I had the same idea a long time ago when i added the led strip code but never got round to it.

hydra, 2015-04-28:
Satellite count could also be blinked out too.

skaman82, 2015-05-13:
That would be great!

multimartin, 2015-11-23:
Any progress on this one ? - just the "lock" status would be extremely helpful - blinking or similar...

meup, 2016-02-27:
I would also very much like to see this feature, please!

rvbcrs, 2016-04-30:
Hi, please this should really be a default function for the led strip! If I want to use the GPS function there should be a way to know what the status is. Best would be through led strip. Is there any progress on this?

gaelj, 2016-05-01:
Patience....

skaman82, 2016-05-11:
thanks, closing

closed 2016-05-11 by skaman82