cleanflight/cleanflight #711: Update Buzzer documentation

heathicus, 2015-04-02:
Include a description (or even links to sound files) of the various beep codes and their meanings.

JohnieBraaf, 2015-04-02:
Not really solving the issue, but this is a interesting read of potentially upcoming functionality. https://github.com/cleanflight/cleanflight/pull/669 it might be done in one go.

ethomas997, 2015-04-17:
There's an updated PR:   Port of refactored beeper code 181_1 #730 

--ET

payne92, 2015-05-14:
I'm working on documenting this in more detail.  Any thoughts on the best way to render buzzer patterns visually in Github markdown?

What about Unicode?

&#8212;  

or 

&#65293; 

or 

&#65293;&#65293;&#65293;

or

&#9608; &#9608;&#9608;&#9608;  (I like this one the best)

JohnieBraaf, 2015-05-14:
Sweet, I like the last one.

hydra, 2015-09-30:
I like the last one too, gotta make sure it renders in the PDF version of the manual though.

payne92, 2015-10-01:
Excellent point, I will keep that in mind.

closed 2017-02-25 by hydra