cleanflight/cleanflight #712: When rotating FC 90 degrees in yaw axis, pitch and roll are reversed

tr3topflyer, 2015-04-02:
I usually rotate my FC 90d in the yaw direction to make the USB more accessible.  Today I discovered that when you do this the pitch and yaw are reversed as far as the board alignment goes.  I found this because I was adjusting for the fact that I'm using 10d tilted motor mounts.  In order get the animation to show the correct attitude I had to adjust roll -10 when yaw is set to 90.  This isn't a big deal except now I'm wondering if PIDs and Rates are reversed as well.  If I want to set my pitch rate to .5 do I need to do it in the roll? and visa verse?

grahamwho, 2015-04-27:
Anyone else have a comment or confirmation on this?

orgua, 2015-04-28:
i can not confirm this. i use the naze32 with yaw +90 adjustment (clockwise rotation, usb-port is on the left now) since CF 1.7.0+ and everything works as it supposed to do. There is one exception regarding headfree-mode, but i don't know if it worked before the yaw-adjustment.

tr3topflyer, 2015-04-28:
It is very easy to confirm.  Just set yaw to 90 and then add -20 in the pitch direction.  View the animation and you will see that the 20 degrees is in the roll instead of pitch direction.  I just tried it with 1.8.1.

https://onedrive.live.com/redir?resid=17ABFC1059650891!1274&authkey=!AOfEAGlkxQ-0yjY&v=3&ithint=photo%2cjpg

https://onedrive.live.com/redir?resid=17ABFC1059650891!1275&authkey=!AFeYKspeDsqe8fs&v=3&ithint=photo%2cjpg

MassiveOverkill, 2015-05-04:
This has already been reported:  https://github.com/cleanflight/cleanflight/issues/272

closed 2015-05-05 by thenickdude

hydra, 2015-05-05:
See #8

joyvel, 2018-03-27:
Hi, I'm new to using the threads on github. I have the same problem of inverted roll and pitch when I performed a 90-degree rotation on my FC. I'm trying to take off in manual and the quad flips, all calibration has been made, I'm using a SPracing F3 and the latest iNav to date. Can anyone help me?