cleanflight/cleanflight #713: Full duplex (dual function) UART port

JohnieBraaf, 2015-04-02:
As enhancement, would it be possible to use a UART port in 2 ways:

RX pin: connected to receiver (SBUS, PPM)
TX pin: connected to minimiosd (MSP)

For example on the Sparky board in this sense, PWM6 (PA2 which is UART2_TX) could be used for this purpose. You effectively save 1 UART port.

cleaflight-configurator 0.1.63.0, new PORT tab could be suitable for configuring with a check in the right boxes.

thenickdude, 2015-04-02:
OSDs require a bidirectional port (MSP protocol requires that you ask for the data you want, and OSDs even change settings over MSP which will always require a bidirectional connection).

JohnieBraaf, 2015-04-02:
Hmmm, ok gonna close this one then.

closed 2015-04-02 by JohnieBraaf