cleanflight/cleanflight #714: Add docs on why CF might not be arming

dkisselev, 2015-04-02:
Going off of this reddit thread:

https://www.reddit.com/r/Multicopter/comments/315umn/switch_from_base_to_cleanflight_wont_arm_anymore/

It looks like there's some confusion about which situations might cause CF to refuse to arm, and people mostly guess at what could fix it.

Could we put together some points in this thread that might cause arm issues? (and if I/someone has time, we'll write the main points into the docs folder)

cmavr8, 2015-04-14:
This would be nice to have, and I think it will be easily completed if you join the IRC channel (#cleanflight on freenode). Many people know a lot of stuff there.

The only condition I know of is if the board is not flat enough. There's a max inclination angle that's tolerated.

So, please come to the channel and we'll figure it out.

hydra, 2015-04-16:
Please create a PR that adds a trouble shooting section to RC controls.  Use the source for reference.

timkostka, 2015-04-17:
I ran into this problem and the issue turned out to be the failsafe was triggering because the throttle was at 984, one below the default threshold of 985.

A calibrated Taranis X9D plus produces values between 984 and 2015.  I think a lower default minimum value makes sense.  Why not just set it to 900, way below anything a reasonable TX would output?

Interestingly, this also made the quad unable to be armed more than once per power-on.  I'm still tracking down the source of that, but it seems to be caused by something in the failsafe logic.

hydra, 2015-04-17:
See #781 and #618

closed 2017-02-25 by hydra