cleanflight/cleanflight #719: pid_at_min_throttle incompatible with retarded_arm

thenickdude, 2015-04-03:
With `retarded_arm` set to 1, the craft is armed with zero throttle and full *roll*. This is incompatible with `pid_at_min_throttle` because it causes the craft to roll over on the ground (the pid_at_min_throttle implementation ignores yaw on the ground but doesn't currently ignore roll when retarded_arm is enabled, so it's not compatible with retarded_arm).

hydra, 2015-04-07:
Does anyone actually use 'retarded_arm'?  Can we delete it?

iacopo, 2015-04-07:
That’s a very timecop-sy thing to say :)
Honestly I get no mode-2 pilot will ever use retarded_arm, but as a mode-1 pilot I find it extremely useful to disarm using a single stick instead of two.
That said, I think I will leave retarded_arm off and start plaing with pid_at_min_throttle, because it looks very cool to do stunt at 0 throttle!

hydra, 2015-04-07:
ahh yes, mode 1 people.  that makes sense now :smile:

iacopo, 2015-04-07:
so many times I thought about switching to mode 2, but it was too late! Like trying to become left-handed.

hydra, 2015-04-07:
Current solution is to disable `pid_at_min_throttle` when `retarded_arm` is enabled.

Please feel free to open a PR with a better solution.  I think this is acceptable for now and won't result in any suprises for mode-1 users.

closed 2015-04-07 by hydra

nebbian, 2015-04-09:
It's useful for tricopter pilots who don't want their tail rotor to hit the ground when disarming.  This happened to me a lot when I was flying a tricopter.  I always rolled my eyes and looked longingly back at my KK2 which never had this problem, it is smart enough to realise that vectored thrust is only useful when there is thrust to vector.

commit caf253e4 (2015-04-07): Automatically disable `pid_at_min_throttle` when `retarded_arm` is
enabled.  Closes #719.