cleanflight/cleanflight #723: Autotune doesn't work and never has worked, please remove it

thenickdude, 2015-04-03:
The basic algorithm of autotune is this:

Rotate to a target angle and stop suddenly
Measure the oscillation at the peak
Use this oscillation to decide on new ROLL and PITCH P, I & D terms
Repeat to test out the new PIDs
At the end of testing, set YAW PIDs based on the ROLL PIDs

There's just a small problem there: In order to rotate to a target angle for testing, AutoTune uses the accelerometer-based "angle" controller. Even when tuning in Horizon mode, because the Rx sticks will be centred, Horizon mode will be identical to Angle mode. And Angle mode **doesn't use the PITCH or ROLL PIDs at all**, those are only used by the Gyro controller which AutoTune doesn't fly with. In fact, apart from on the yaw axis, Angle mode only uses the LEVEL PIDs which AutoTune never modifies.

So in other words, AutoTune never measures the effect of the PIDs that it is tuning. So it will never converge on a solution, and one iteration is as "good" as a thousand.

In order to actually tune the ROLL & PITCH PIDs, AutoTune would have to do its tuning in Acro mode, so that those PIDs were really being used. And that would require its "rotate to a given angle" structure to be rewritten to return target rotation rates to the PID controller instead of returning target angles (i.e. rotate at a fixed speed until the angle reaches the given target, then start returning 0 for the target rate).

If you look back at BradWii, where AutoTune came from, that project *only uses an Angle controller*, which actually uses those ROLL & PITCH PIDs, so AutoTune was completely sensible there. But not on Cleanflight.

Please remove it since it never "tunes" anything and is actively dangerous. It could be added back in if it was ever rewritten.

acceleroto, 2015-04-03:
Agree. Yay for flight test proof of the existing autotune not working. Let's ditch it and move on to something that makes sense for Cleanflight.

hagisbasheruk, 2015-04-03:
I am also in total agreement , can wen not use the blackbox feature to gather data for an autotune solution ?

acceleroto, 2015-04-03:
An offline tune is definitely possible. If the autotune algorithm was replaced with a good series of canned inputs (designed specifically for system parameter identification), it'd be pretty easy to do a system identification offline using Blackbox data and then generate PIDs from that.

ledvinap, 2015-04-03:
@thenickdude : It's not exactly true that roll & pitch is not used. 
Angle PID is supplying P and I term from desired angle. But gyro PD controlle(with roll & pitch PD parameters and possibly dynamic attenuation) is still active with zero setpoint.

But I agree that autotune is totally broken,

ledvinap, 2015-04-03:
relay PID tuning may be interesting option: 
http://www.controleng.com/search/search-single-display/relay-method-automates-pid-loop-tuning/4a5774decc.html

thenickdude, 2015-04-03:
ledvinap: With the multiWii PID controllers? The Gyro PID terms aren't used at all with those in Angle mode.

closed 2015-04-03 by thenickdude