cleanflight/cleanflight #724: Beeping on "Save" in PID Tuning - UI

Petfrogg, 2015-04-03:
While connected with bluetooth and when pushing "Save" the model should be able to give some feedback. The usability of this is best depicted when doing PID changes using bluetooth.

I suggest 2 impulses (or more if it collides with any already in use) sent to the beeper when saving or refreshing while using any wireless connection.

thenickdude, 2015-04-03:
You mean to confirm receipt of the command to save? In theory the Configurator can already react to that, because when an MSP command is received, it replies with a "success" message. When no "success" message is received, the Configurator should print a timeout error message to the log at the top.

ethomas997, 2015-04-05:
Seems like when I'm connected via bluetooth and the connection is wonky, when I do a save (that fails) the Configurator shows no response message at all (no "EPROM saved" or error message).  A confirmation beep on save would be a plus, IMO.

--ET

hydra, 2015-05-15:
This is a GUI issue, please open something appropriate in the other issue tracker.

closed 2015-05-15 by hydra

Petfrogg, 2015-05-16:
I thought the beeping is something that relates to cleanflight and not the configurator. Since it is the board that needs to do the beeping shouldn't it be here then? Please direct me on where to post this.

thenickdude, 2015-05-17:
IMO this is a Configurator bug. The configurator already (should) know if
its command has made it safely to the FC or not (by listening for a serial
ack), so there's no reason why we need the FC to confirm it separately. The
configurator needs updated so it properly communicates the success status
to the user (assuming it doesn't already, check the error log at the top).

thenickdude, 2015-05-17:
Also note that the FC cannot beep unless the flight battery is connected
(by Naze32 design) so we couldn't add a reliable beep anyway.