cleanflight/cleanflight #726: 1.8 flash cc3d board to cleanflight

eranme, 2015-04-04:
Got a new board from fpvmodel.com flashed with. Cleanflight 1.8 stable can not connect via usb i am using win 8.1 64  the led light on the board is orange no issues in devices on the pc.  The pic make a sound that a device is connected no error but the software can not find the board

dascons, 2015-04-04:
You need to use some drivers if you don't use open pilot:
http://www.st.com/web/en/catalog/tools/PF257938

jamieharwood, 2015-04-04:
I have the same problem on my Mac. Is there a similar driver I need to install?

eranme, 2015-04-04:
Thank you dascons i installed the driver as explained and no luck....

eranme, 2015-04-04:
i tried connecting it to my Mac no luck...

eranme, 2015-04-04:
jamieharwood
click on the link https://wiki.openpilot.org/display/WIKI/Bootloader+Update
it will help you to revert back 
i will wait for a fix and then try to flash back to cleanflight

tricopterY, 2015-04-04:
Power up the FC using its USB port via your PC/Mac. If you have the driver installed your computer would detect the FC and allocate a virtual com port. Then open the Chrome based CleanFlight Configurator. 
Configurator should find all the virtual com ports and populate a dropdown list. You should select one com port and attempt the connection. If it fails, close the Configurator, then restart it and try other reported com port(s) from the dropdown list.

Telnoi, 2015-04-11:
both the usb cable and the lipo have to be connected. Without lipo, my cc3d will not connect either.

FranzAr, 2015-04-11:
I've made a small Video on installing Cleanflight on CC3D. Read also the description and comments. Cleanflight on CC3D is a bit tricky to handle, but if you follow the instructions all works fine.
https://youtu.be/_9zxu2vmdlY

closed 2015-04-14 by hydra

Indy6, 2015-10-19:
Hi guys,
As I understood, on a mac with a CC3D, I have to plug a battery? So that means I can't synchronise my ESC with Cleanflight since the first step is to put the motors to the max without the battery, isn't it?

FranzAr, 2015-10-19:
No problem, first connect with battery plugged. Then unplug battery, the USB connection will remain active! Go to Motor Tab and synchronize as usual.

Indy6, 2015-10-19:
Thanks FranzAr, unfortunately it's my stupid day. Trying to synchronize my Emax Simonk ESC with the throttle, I made a mistake and set the ESC to bidirectional. Now it seems I can't enter the program mode, the security doesn't allow me to start throttle up... I'm stuck... ;( thanks anyways

A-Z-que, 2017-01-31:
hello i have a cc3d and i connected the battery and openpilot is discontinued and my CC3D wont connect to cleanflight