cleanflight/cleanflight #728: Autotune causes altitude climb when thrust to weight ration is high

NullVoxPopuli, 2015-04-04:
I tired using auto tune on my warpquad, and as it was making adjustments, it was climbing. 
I have yet to complete an auto tune session, cause of how high it gets :-\

acceleroto, 2015-04-04:
There's lots of development talk about autotune right now.  The short story is that it doesn't work.  My strong recommendation is: Don't use it.

NullVoxPopuli, 2015-04-04:
Oh ok. cool. thanks :-)

(I'm fine flying in acro) :-)

acceleroto, 2015-04-04:
You might want to try G-Tune then.  It's still in development, so be careful.

illuxion, 2015-04-05:
autotune with this code is EVIL.  It's not only the problem of high thrust to weight, but that it ignores all failsafes, or at least it sets the PIDs so whacky what the failsafes are ignored.  I spent 2 weeks building and testing a mini quad bros kit, replacing the acro with a full Naze32, adding sonar, and a few other goodies(wish I would have added GPS).  I had been honing my PIDs and had a pretty well behaved quad with a little bit of Z and yaw issue so I decided to hit auto tune.  It did a few normal back and fourths then absolutely lost it's mind and went skyward.  I hit disarm but it continued up so I turned off my taranis, but it STILL continued up.  Failsafe and disarm should take priority over EVERTHING else, but they don't so I'm out a few hundred bucks.

anees042, 2015-06-20:
when ever i enable auto tune, my quad falls from the sky, the altitude normally is about 5 meters. any recommendations?

acceleroto, 2015-06-20:
Ummm, don't use autotune.  The docs say not to use it right now.

hydra, 2015-10-06:
autotune has been replaced with G-Tune.

closed 2015-10-06 by hydra