cleanflight/cleanflight #731: Option RX_SERIAL not saves via GUI

azaitsev, 2015-04-04:
After reboot no receiver mode selected

thenickdude, 2015-04-05:
Try configuring the port on the ports tab for Rx serial before enabling the feature.

azaitsev, 2015-04-05:
It helps. But it's not user-friendly.
Also no power on my RX via SBUS.

azaitsev, 2015-04-05:
My bad. Forgot to connect battery :)

NullVoxPopuli, 2015-04-10:
I second this. 
Configuring RX_SERIAL is not as intuitive as it should be.

I have a suggestion, which I'll post to the configuration issues: 
It should be possible to select "popular" receivers, and have the configurator set whatever parameters it needs to, so that there is less command line / apparent not saving due to feature conflicts.

hydra, 2015-04-16:
This is a configurator issue. closing. please open in the configurator tracker.

closed 2015-04-16 by hydra

Propitiation, 2016-01-30:
where is the configurator tracker?

thenickdude, 2016-01-30:
https://github.com/cleanflight/cleanflight-configurator/issues