cleanflight/cleanflight #732: disarm_kill_switch Dangerous

tracernz, 2015-04-05:
I have received reports of dangerous situations where quads have flown away when competing at a local mini-quad racing event over the weekend. All three of the quads involved were setup using an RC channel for arm/disarm and were upgraded to CF 1.8 prior to the weekend. What occurred was that the pilots flicked their switches to the disarm position (equivalent to throttle hold for us heli pilots) and the quads continued flying at the previously commanded throttle and attitude rather than disarming and crashing as expected. I see this as an extremely dangerous behaviour which shouldn't even be possible.

I was given the NAZE board off one of the quads to try and figure out what is happening. The cause is that the disarm_kill_switch setting was set to 0, i.e. the quad will not disarm unless at zero throttle. This is an extremely dangerous idea and I really can't understand why it would even be possible? 

The root of the problem is that the pilots backed up their configurations using the configurator before upgrading (the backed up config was from API version 1.5 on the sample I received). The boards were then flashed with CF 1.8 and the configuration restored. As the disarm_kill_switch was added to the MSP command set in between these API versions it wasn't present in the old backup file, I assume it defaulted to 0/disabled when restoring the backup. The resulting behaviour was completely unexpected by the pilots involved and could have easily resulted in injury.

If this setting/feature must be kept then at least it should default to 1 when restoring an old backup file to avoid such incidents in the future.

sjtrny, 2015-04-05:
The quad will not disarm unless at zero throttle so that you don't accidentally disarm it mid flight.

ScottySpoon, 2015-04-05:
I understand the theory behind why you might want it, to avoid accidental disarm, but in practice its a terrible idea.... disarm should be disarm.... no matter what.... its a safety thing

Daveskie, 2015-04-05:
Maybe an option to choose between just a disarm switch or a kill switch would be better. Personally I only want it to work when I don't want the motors spinning up. 
I do however think that Failsafe should kill the motors is set to do so.

tracernz, 2015-04-05:
@sjtrny, I guess that's why it was added but it seems to me that's treating the symptom rather than the problem (i.e. the operator is not in control of his radio)?

tracernz, 2015-04-05:
Another example of why it's a bad idea: https://github.com/cleanflight/cleanflight/issues/728#issuecomment-89743376

sjtrny, 2015-04-05:
@ScottySpoon Except when you accidentally disarm over people or buildings.

digitalentity, 2015-04-05:
Flying over people is dangerous by itself. What if a motor fails? If I
loose control of a copter, I would attempt to minimize consequences and
having a option to disarm instantly is a must have.

-- 
С уважением, Константин Шарлаимов

Regards, Konstantin Sharlaimov

tracernz, 2015-04-05:
@digitalentity, agreed, I hadn't considered that use case (flying over people) because I find it unacceptable to begin with.

ScottySpoon, 2015-04-05:
Flying over people........ YOU SHOULD NEVER FLY OVER PEOPLE..... so many thing can go wrong at any time, and the one thing that should work every time is the switch that stops the spinning knives....

sjtrny, 2015-04-05:
I agree with you all that disarming and kill switches should remain as features. However not every user wants this so the choice needs to be provided to them. So I'm not really sure what you're arguing about.

digitalentity, 2015-04-05:
User must have a choice, but defaults should be the most safe ones. And the
safest default in this case would be an always working kill switch.

-- 
С уважением, Константин Шарлаимов

Regards, Konstantin Sharlaimov

sjtrny, 2015-04-05:
Yes which is exactoy what I said in my first post.

ethomas997, 2015-04-05:
The 'disarm_kill_switch' CLI setting has been in the code since at least Cleanflight v1.0.0 (2014), and its default value is '1'.  So if it wasn't enabled then most likely the operator disabled it.

--ET

tracernz, 2015-04-05:
Yeah, it was available as a CLI variable but it has just been added to MSP and the configurator in the last release which is where this issue came from. :wink:

ScottySpoon, 2015-04-05:
i can confirm that if you back up ur config file in the configurator (previous version) then load the new firmware and reload your settings, it sets the kill switch disarm to 0 as default

tricopterY, 2015-04-05:
As per the docs - 'Setting to 0 reverts to the old behaviour of disarming only when the throttle is low. Only applies when arming and disarming with an AUX channel'. But default has always been 1, which to me is very odd, as you can accidentally disarm mid flight.
The bug is:
```C
var ARMING_CONFIG = {
    auto_disarm_delay:      0,
    disarm_kill_switch:     0      //should be 1 for default
};
```

hydra, 2015-04-07:
This is a configurator bug, not firmware bug. Closing.  Someone please create a new issue in the configurator issue tracker and summarise, give a link to this too.

closed 2015-04-07 by hydra

tricopterY, 2015-04-07:
@hydra - should the above code then be:
```C
var ARMING_CONFIG = {
    auto_disarm_delay:      5,
    disarm_kill_switch:     1      
};
```

seanmavley, 2021-01-24:
> @ScottySpoon Except when you accidentally disarm over people or buildings.
> 
> @tracernz People make mistakes all the time, it's not hard to accidentally flick a switch. I accidentally disarmed mid flight last Sunday.

Thank you. I'm not sure what's so hard to understand about this. There's 2-step arming sequence. Why can't there be 2-step disarming sequence?

I'm digging through archives, only to learn this 2-step disarming sequence i.e 1) flip disarming THEN 2) throttle to 0 to complete disarm sequence.

I damaged a motor and battery (beyond redemption) today, all because I accidentally disarmed mid-air. 

The logic of instant disarm is like saying turn off car engine when about to hit something. You rather apply brakes (in this case, disengage throttle) THEN turn off engine (complete disarm)

From what happened today, there was no time to re-arm and catch the drone because it tilted in an angle which wouldn't allow re-arming. But with a 2-step disarm sequence, I could have had time to 
 - re-arm the drone (because throttle was still up) and continue flying OR
 - cut throttle and come crashing to ground like a stone (assuming it was true emergency)

None of this new "feature" makes sense to me.

haslinghuis, 2021-01-24:
instant disarm is indeed some form of panic button now. Should be tied to zero throttle or using `auto_disarm_delay` and `disarm_kill_switch` - didn't try if this is still functioning.