cleanflight/cleanflight #736: Cold Temp Arming

ntgsteph, 2015-04-05:
I fly a lot in cold temps and the acro naze32 will not arm at sub zero temp. I can get it to arm inside a vehicle and then fly without issues, but if a i change the battery it will not re-arm until it has been warmed up again. This could possibly be a CLI command adjustment bu not sure what to adjust. Thanks for any help.

ledvinap, 2015-04-05:
Try increasing small_angle. ACC reading are probably affected by temperature and small angle test is checking ACCz only, without normalizing ACC vector. The test will always fail if ACC vertor length is too small.

ntgsteph, 2015-04-05:
I increased the small_angle variable from 25 to 40 and the cold weather arming issue went away. Thanks a lot for the help!

closed 2015-04-05 by ntgsteph