cleanflight/cleanflight #737: Wrong SMALL_ANGLE  calculation

ledvinap, 2015-04-05:
SMALL_ANGLE condition is checked as

- https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/imu.c#L92
```
smallAngle = lrintf(acc_1G * cosf(degreesToRadians(imuRuntimeConfig->small_angle))); 
```
- https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/imu.c#L274
```
    if (EstG.A[Z] > smallAngle) {
        ENABLE_STATE(SMALL_ANGLE);
    } else {
        DISABLE_STATE(SMALL_ANGLE);
    }
```

With small `cosf(small_angle) - 1` this calculation is very sensitive to accelerometer gain, causing problem when flying in freezing temperatures (acc is calibrated inside, you can't arm after a while when IMU cools enough). It if very confusing for users.

Basing the test on `inclination` may be quite easy fix and should not break anything ..

hydra, 2015-04-05:
I have noticed this a few times on a couple of the boards I have, some boards seem more sensitive than others.  Some do it even at mild temperature whilst others are fine even on the tops of mountains after being calibrated indoors.

A temporary workaround seems to be to rotate and twist the quad around on it's axis' and then leave level for a moment and the small angle value drifts back into range - might just be the sensor warming up though...

Do you have a fix?

ledvinap, 2015-04-06:
@hydra: I'll look into it eventually ... Using `inclination` is probably enough.

It is strange that rotating quad helps (I can confirm that). It's not warming up - few times I was flying few minutes, but cannot arm again after landing.

Maybe the complimentary filter is too slow, but I remember spending few minutes pondering what's wrong. Or there is some rounding issue (not likely, but gyro_cmpf_factor==600 is almost half of available float mantissa bits).

hydra, 2015-04-06:
@ledvinap yeah, I fitted an OLED screen to help debug this one one quad in the field and thats when i noticed that the value eventually drifts back to the right value after moving the quad around.

I have had reports from other users about this, it needs fixing.

hydra, 2015-04-11:
Any chance you could look into this one a bit @ledvinap ?

ledvinap, 2015-04-11:
IMO this is enough:
```
if(ABS(inclination.values.rollDeciDegrees) < imuRuntimeConfig->small_angle * 10
  && ABS(inclination.values.pitchDeciDegrees ) < imuRuntimeConfig->small_angle * 10)
   ENABLE_STATE(SMALL_ANGLE);
else
   DISABLE_STATE(SMALL_ANGLE);
```

There may be problems if small angle is large; the restricted region will be pyramid-like. But it does what it is intended to do ...

closed 2017-02-25 by hydra