cleanflight/cleanflight #739: airspeed / wind speed sensor support

NullVoxPopuli, 2015-04-06:
I think it would be pretty cool to support an air/wind speed sensor to see how fast the aircraft is moving / keep track of the max speed.
I don't know much about the sensors, but something like this http://store.3drobotics.com/products/airspeed-kit-with-mpxv7002dp (the only sensor I could find in a 5 second search) wouldn't work for multi-rotors due to how the angle of the aircraft changes during fast forward flight, which would reduce the air-intake to the tubes.

dkisselev, 2015-04-06:
If you have GPS, that'll give you the ground speed, but I'm not sure how you would get air.

You could stick a tilt servo on it with that standard air speed sensor: https://www.youtube.com/watch?v=nl6I00yCfKY, cleanflight has a configuration option that will keep the servo level for you as it flies.

hydra, 2015-04-07:
Duplicate of #529

closed 2015-04-07 by hydra