cleanflight/cleanflight #740: Virtual Current Sensor documentation unclear

dkisselev, 2015-04-07:
I'm trying to setup the virtual current sensor, but the docs aren't quite clear on how to do it.

```
Adjust current_meter_scale until the mAh draw reported by Cleanflight matches the charging data given by your battery charger after the flight (if the mAh draw is lower than reported by your battery charger, increase current_meter_scale, and vice-versa).
```

By the time you get home to the charger, the board's been powered off so CF's mah count it lost and you can't tweak it.

I'd like to help improve the docs to make the setup more clear, but don't know what the process should be.

If anyone's done it and has a step-by-step worked out, I'd love to hear it, otherwise I'll poke around and post my finding here (and update the doc if I get somewhere conclusive)

hydra, 2015-04-14:
Perhaps check the closed PR's to find the author of the virtual current meter code and ask them, you can just mention their github username in a comment.

dkisselev, 2015-04-18:
Here's my data for a few packs, it looks like the current sensor is actually really accurate even with no configuration on some builds (I have a 450). Virtual is what CF reported at time of flight, charger is how much I put back into the pack.

Virtual: 1346mah  
Charger: 1219mah

Virtual: 1266mah  
Charger: 1185mah

Virtual: 1260mah
Charger: 1148mah

Just leaving this here for anyone's reference. Gonna play around with the the code to see if I can get it even more accurate.

handsomejackuk, 2015-04-24:
nice one i need to monitor this too....

payne92, 2015-04-29:
Could someone here eyeball doc edits #820 to see if they address the questions here?  Thanks!

digitalentity, 2015-04-29:
I just did so, looks much more clear and understandable to me.

dkisselev, 2015-04-29:
@digitalentity We need to actually test a full cycle to make sure the instructions/calculations actually work (they do in Theory, but a live test would be best).

I've just been too busy to go out and fly twice since the discussion started so haven't been able to charge my batteries and test the real usage after making my changes.

payne92, 2015-04-29:
@dkisselev I agree, but I worry that there's a dependency on the accuracy of the particular battery charger that may make it work for some and not work so well for others.  

My preference is to disclaim that and highlight that this is an *approximate* method, better than nothing, but not a full substitute for a current meter.

digitalentity, 2015-04-29:
@dkisselev, I fly with virtual current sensor - don't have needed hardware yet. Calibrated it using an ammeter and now it is pretty reliable. However I agree that we must state that it is an approximation of actual current draw.

payne92, 2015-04-30:
Good comments, I updated #820 accordingly, adding:

"Note:
+ This method depends on the accuracy of your battery charger; results may vary.
+ If you add or replace equipment that changes the in-flight current draw (e.g. video transmitter, 
+ camera, gimbal, motors, prop pitch/sizes, ESCs, etc.), you should recalibrate."

PR #820 (2015-05-06): Battery.md:  clarify current_meter_scale
fixes #740 

@dkisselev  Per your comments on #814 about calculating `current_meter_scale`, could you eyeball and see if this addresses your comments?
closed 2015-05-06 by hydra