cleanflight/cleanflight #742: Latest fw CF with cc3d chanel5  issue

ultrojo, 2015-04-08:
cc3d board
My RX Futaba 7008SB
sbus did not works
if use pwm works only for chanel AETR, Ch 5 for Mode not works

hydra, 2015-04-14:
Please provide configuration dump

hydra, 2015-04-14:
Likely you still have VBAT turned on (it's on by default).

Please re-open if there's still an issue.

closed 2015-04-14 by hydra

hydra, 2015-04-14:
This commit for v1.9.x may help future users - 436d2ed5917a3f4754e5a2b4e0ad33a8e224b28a

hydra, 2015-04-14:
I verifed 1.8.0 and 1.9.0 (development) using the pinouts described in the documentation using PWM with and without VBAT enabled and everything works as expected.   No issue found, please ensure your connections and settings are correct before opening issues otherwise I have to spend my time testing things instead of making the code better for you all.