cleanflight/cleanflight #744: Autodisarming failsafe after flight

HV40, 2015-04-08:
hi,
This is feature request following me messing up in the field, I disarmed my quad after the flight and then switched off my transmitter before powering off the quad => this triggered the failsafe as designed, motors spinning... no harm done though.

I though a way to prevent this would be to disable the failsafe after a certain period of time after disarming the board, if the channel that is checked for failsafe is still valid.

Maybe it's a bit confusing to understand, let me try to put it simpler :
Disarming board => start timer => time elapsed (say 1sec) => other channels valid ? Yes => disable failsafe.

Failsafe being always armed when the board is armed.

The reason for the timer is to dissociate this scenario from the actual failure of the tx/rx where you could have the failsafe switched detected as disarmed when the link is actually failed.

I believe this would render the board more safe to idiots like me :)

dascons, 2015-04-08:
Makes sense to me :)
I think this should be implemented as I also have been in such situation

tricopterY, 2015-04-10:
Not sure if the above scenario is possible - failsafe gets activated, only if the craft is Armed and elapsed time of not receiving valid RC data. Please remove the props and reproduce your steps, and report your findings.

hydra, 2015-04-16:
Probably fixed by this:

https://github.com/cleanflight/cleanflight/pull/781

hydra, 2015-04-16:
Specifically this:

https://github.com/cleanflight/cleanflight/pull/781/files#diff-2802db4b528c0ac6414bf6889ea3c4fdR164

rscarawa, 2015-04-25:
I will look at 781 to see if that improves failsafe for what I am looking for.  I would like the failsafe to suspend operations if either the board is disarmed or the throttle is at idle.  The failsafe should resume if the board is rearmed and throttle exceeds idle.  Here is the problem situation.  After my flight or test, I immediatly go to unlook my quad battery.(Iknow this is a bad practice) but if the tx gets too close to the rx, it floods it and failsafe goes off.  This is common with FSSky systems if the tx gets within 3 feet of the rx.  Then it is possible to the motors to spin up while I am trying to unhook my battery.  Has this been covered in 781?

hydra, 2015-05-15:
@rscarawa Please re-open if this is not fixed in current master.

closed 2015-05-15 by hydra