cleanflight/cleanflight #746: Async ESC initalization with Oneshot, maybe throubles with serial

Exhdous, 2015-04-08:
Hi

I've found two interesting things, they might not cause an issue at all, but knowing about it can never hurt!

So the first thing I realized, when Oneshot is enabled the ESC-s are no longer intializing in sync, which makes sense, as before oneshot they were updated on a given rate and I guess the whole initialization was done between to updates, so oneshot only brings out the sequential manner of initialization.
It does not seem to cause any issues with flight, but should be added to the documentation at least, I've spent ours trying to find the cause of the asnycronisity, others might be wondering as well.

On the other hand, and this will be the more interesting one, when I connected the USB to the board, the initialization of the 4 esc-s was perfectly in sync, this tells me that either the first issues I described is a bug, or connecting the USB actually disables the Oneshot protocol, which can be problematic if someone uses OSD or something else that uses the serial port just like the USB does.

As for environment, Im using a full naze32 board, running Cleanflight 1.8, the ESC-s are Hk blue-s with Blheli 13

Keep up the good work! :)
Exhodus

modem158, 2015-04-09:
I'm going to upgrade my ESC's to Blheli 13 this weekend.  According to the docs they should recognize OneShot automatically.  I'll see if I can verify this behavior on another setup.

hydra, 2015-04-14:
We'll deal with this in #546

closed 2015-04-14 by hydra