cleanflight/cleanflight #748: Sparky RSSI ADC

TexasFPV, 2015-04-09:
I flashed the latest build onto my Sparky board.  The configurator allows enabling the RSSI ADC features, but the documentation doesn't indicate which pin to send the RSSI feed from my Frsky receiver.  I tried to deduce the correct pin from the source code, and I couldn't find any RSSI support in the Sparky target code.  Is this a bug in the configurator, or just an unimplemented feature?

DaTwo, 2015-04-09:
Go into the receiver tab. You will see a drop down for "RSSI Channel" This is the servo channel you will be using for RSSI. Since the Sparky only has serial RX input you must send back the RSSI data from the transmitter. If you are using a Taranis this is easy. Just map the RSSI to a channel. Set the weight at 200 and the offset at -100 and the RSSI should match whats on the transmitter. Hope this helps.

TexasFPV, 2015-04-09:
This is helpful.  I suspect that ADC RSSI can be implemented on the Sparky however.  For example, the Taulabs software allows ADC input on the Sparky PWM out 8,9, and 10 pins.  Currently in Cleanflight pins 8 and 9 are used for monitoring battery and current, so the target file could probably be modified to allow analog RSSI monitoring on pin 10.  Though as you point out, this isn't really necessary if RSSI is mapped to a channel on the Taranis.

JohnieBraaf, 2015-04-12:
Currently pin 8 (amp) and 9 (voltage) have some ADC measuring. Maybe it can be implemented on pin 10. But it was mentioned that it uses a different ADC channel here https://github.com/cleanflight/cleanflight/pull/635 and here https://github.com/cleanflight/cleanflight/issues/523 .

closed 2017-02-25 by hydra