cleanflight/cleanflight #749: Motors Not Run

kingstien, 2015-04-10:
I flashed the cleanflight v1.8.1 into my CJMCU MiniCopter and all adjust had be done.I run the motor test the motors work well,but it not work when i use the transmiter.
I think it's not disarmed or something else have problem.(I have checked the MOTOR_STOP and set delay to 1 second)

add1ct3dd, 2015-04-10:
Check the channel layout is correct...

kingstien, 2015-04-10:
The channel layout is correct.

add1ct3dd, 2015-04-10:
And have you verified it in the receiver tab to make sure all endpoints etc are correct?

If you arm it does the LED light up? More info needed.

kingstien, 2015-04-10:
LED1(Yellow),LED2(RED),LED3(YELLOW)
Only LED3 light up,when i hand it up and move,the LED2 flash.
 all endpoints are correct.

kingstien, 2015-04-10:
[image]
[image]

kingstien, 2015-04-10:
The v1.7.2 and v1.8 same

add1ct3dd, 2015-04-10:
Hmm you're right I don't see why it wouldnt work - you need to trim your roll/pitch/yaw so they are actually centered at 1500 though :)

kingstien, 2015-04-10:
May be my CJMCU is a new version,there's something different from before verisons.I bought it recently.

hydra, 2015-04-10:
I just seems like you're not arming it.  A value of 1 second is very short.  reset it back to 5, use the correct stick combination. See manual.  Please re-open the issue if you still have a problem.

kingstien, 2015-04-10:
I tryed.It not work.

thenickdude, 2015-04-11:
Is 1114 the lowest that your throttle goes? If so, it doesn't go low enough to perform stick commands (min_check controls that threshold, and it defaults to 1100). Fix your transmitter's endpoints so that all your channels span the range 1000 to 2000.

kingstien, 2015-04-12:
Thank you. It works. ^_^

closed 2015-04-12 by thenickdude