cleanflight/cleanflight #750: failsafe not triggered - CC3D + Orange R615X

Telnoi, 2015-04-10:
-Cleanflight v1.8.1
-Configurator v0.63.0
-CC3D FC
-DX6i + Orange R615X connected through PPM

Issue: failsafe is not triggered when all 4 signals drop below the threshold value of failsafe_min_usec (993). 

When I have my transmitter turned on with these failsafe values, the quad will increase throttle to the indicated failsafe_throttle value of 1400.
As soon as the transmitter has been turned off, the motors will stop spinning, even though the receiver tab clearly outputs the ''incorrect'' signal output as described in the wiki.

The Orange R615x's has a programmable failsafe, which allowed me to output the correct values below the failsafe_throttle value as soon as the connection is lost between Tx & Rx. Obviously, that part worked (see screenshot below of the output when failsafe is triggered through the Rx.

Values with Tx turned off.
[image]
Values with Tx turned on.
[image]
Failsafe values.
[image]
Failsafe is enabled.
[image]
ESC idle settings etc.
[image]

hydra, 2015-04-10:
When you switch the TX off, when armed, does the quad disarm?

Telnoi, 2015-04-10:
Seems so yes. The blue light turns off. Green light on the CC3D is solid.
Red light on the Rx is flashing.

Telnoi, 2015-04-10:
Some more information.

-Failsafe is not activated when turning off the Tx. Throttle does not increase to 1400
-Failsafe is activated if I disconnect the Orange R615X/pull the wire. Throttle increases to 1400.

It would seem that failsafe does not trigger with invalid signal values. It does trigger when there is no signal at all.
After having read the WIKI a few times, it would seem that cleanflight's failsafe should work either when there are invalid signal values triggered (based on the failsafe thresshold) OR when there is no signal. 
The Orange RX only supports programmable failsafe, meaning that I can only output pre-programmed signals (in this case, all below the failsafe thresshold). I can not program it to send no signal at all.

Anything I can do to fix this? I love cleanflight and prefer it to OpenPilot (flight characteristics are so much better).

Telnoi, 2015-04-10:
Solution & proposed fix.

Found the solution. Either the behaviour is unintended, or this is missing in the wiki.
When using invalid signal method with a programmable failsafe (hold) for an Rx that does not support ''no signal out'', the failsafe has to be programmed in such a way that ARM will be triggered using for example AUX 1 (see screenshot).

Failsafe will automatically disarm after the defined period through CLI. In my opinion, Cleanflight should ALWAYS keep arm enabled for that defined perior, regardless of the arm state signal provided by the Rx.

ethomas997, 2015-04-10:
It could be tricky to implement making the TX-AUX-switch arming stay enabled because the failsafe changes in the throttle and TX-AUX-switch (disarm) channels will happen simultaneously.  The wiki, though, should definitely be updated to say that the arming channel needs to be set on failsafe for the "auto-landing" function to work.

--ET

Telnoi, 2015-04-11:
Perhaps the armed status can ignore any signal induced de-arming sequence and stay enabled during failsafe detection until the timeout. This seems to be happening when I pull the Rx wire, since that will most certainly not produce the correct AUX value for arming..yet the motors will start spinning.

After the timeout, follow regular procedures (not being able to arm until reboot etc).

hydra, 2015-04-13:
@Telnoi yes, i suspected it might be something to do with the auto disarm code.  Can you confirm by disabling auto-disarm (set it to 0) and turning of the tx while armed?

Telnoi, 2015-04-13:
I will receive a long-range Rx sometime this week which I will have to set up.
I'll let you know the outcome of using auto-disarm 0 instead of ensuring that arm is triggered through a programmable failsafe aux channel value.

P.S. failsafe will trigger after disarming the quadcopter & turning off the Tx. Separate issue, but that should of course not happen either.

hydra, 2015-04-16:
Please see and test this:

https://github.com/cleanflight/cleanflight/pull/781

Telnoi, 2015-04-20:
Sorry, what specifically should I be testing?

hydra, 2015-05-06:
@Telnoi Hiya.

Please grab a recent development build and re-test the failsafe.

Telnoi, 2015-05-07:
Could you point me to a bin file (at least that's how I installed it on CC3D in the past), or will HEX do the job through OpenPilot?

Seems a compiled version is not available of master, no OBJ dir. I might be missing something.

hydra, 2015-05-07:
Builds are available from the build server now. See irc topic or twitter posts for URL.

Telnoi, 2015-05-07:
I tried flashing 1114_69292fc_cleanflight_CC3D.bin, but receive the message in CC that the firmware version is not supported (need api v1.2 or higher support). Any idea?

hydra, 2015-05-15:
@Telnoi Use configurator 0.63.2 which supports API > 0.9.0.

closed 2017-02-25 by hydra