cleanflight/cleanflight #752: Code Metric and Smell analysis?

NullVoxPopuli, 2015-04-10:
I really admire the goal of cleanflight - applying good software engineering principles, etc
But I think it would be awesome to see the current state of existing smells and another static analysis metrics.

Maybe something like http://oclint.org/ could help?
(I don't do much C/C++ programming yet, so I don't know what would be best here. (I'm doing Ruby (personal projects) and VB.net (client project) at the moment))

hydra, 2015-04-10:
Yes, it would be good to add some report generation to the automated Travis builds.

NullVoxPopuli, 2015-04-10:
Travis is great! I use it for my ruby gems.
I can try to set that up, but I figure travis should be in your name, yeah? @hydra

NullVoxPopuli, 2015-04-10:
I think we want this for code coverage: https://coveralls.zendesk.com/hc/en-us

NullVoxPopuli, 2015-04-10:
We might want landscape.io for code analysis? https://landscape.io/github/NullVoxPopuli/ardupilot I'm not sure how well it works yet, but it's going over my ardupilot code now. o.o

dkisselev, 2015-04-11:
Coveralls would be my vote.

I've also seen [Scrutinizer](https://scrutinizer-ci.com) used a lot, but I find that having half a dozen CI badges on a project tends to look bad, so maybe just the two is enough.

NullVoxPopuli, 2015-04-13:
I think meaningful badges make a project look good.

closed 2017-02-25 by hydra