cleanflight/cleanflight #755: Sonar drift due to modifying Rx values

nebbian, 2015-04-11:
I've just been testing the sonar altitude hold, and have noticed that when altitude hold is enabled with only a sonar module, the craft will drift slowly (~2cm/s) downwards, until it is very close to the ground.  This is without touching the throttle.  Modifying the alt_hold_deadband to a much higher value than default stops this drift, despite not touching the throttle stick the entire time that this is happening.

I've been going through the code to try to understand what's going on, and have found the issue.  ApplyAltHold is called every loop, which will modify rcCommand[THROTTLE].  However, processRx (which resets rcCommand back to the actual Rx values) is only called at 50 Hz intervals.  

This means that ApplyAltHold will modify rcCommand[THROTTLE], and on the next run through the loop will notice that rcCommand[THROTTLE] has changed.  If this change is more than the deadband, then it will reset AltHold to EstAlt, which means that the setpoint changes.

I think that changing Throttle RcCommand values to control altitude is the wrong way to go about things.  To my mind, the only part of the code that should change RcCommand values is the part of the code that actually reads the receiver values.  

I'm not entirely sure about the best way to go about fixing this either.  I could:
* Only run applyMultirotorAltHold if rxUpdateAt has changed since the last run through, which is really messy and increases code coupling
* Try to add some more variables to every part of the code to try to separate the rcCommand variables from code that tries to control the craft (eg: Altitude hold, Magnetometer hold, HeadFree mode etc).

How would you go about sorting this out?

ledvinap, 2015-04-11:
Maybe using value from `rcData` instead of rcCommand may improve the situation?

hydra, 2015-04-11:
@nebbian I was testing sonar myself on my new FC the other day and noticed the behaviour you described. Thanks for having a look into it so far.  Maybe try a PR as @ledvinap suggested

nebbian, 2015-04-12:
Pull request created:
https://github.com/cleanflight/cleanflight/pull/759

hydra, 2015-04-14:
The PR is merged, can people please test new development builds and report their findings here.

DIY420, 2015-04-24:
Sorry I'm a newbee here. How can I test this dev build? Is this software available via the normal cleanflight GUI?
I'm very interested in this bug fixing, because without this I don't want to implement the GPS module. Unfortunately I'm not at all a software developer...

hydra, 2015-05-06:
@DIY420 Hiya.  Development builds are available here: http://cleanflight.memoryleaks.org/builds/

Please test and report back.

hydra, 2015-05-06:
@nebbian Do you consider this fixed now?

Ben250, 2015-05-07:
I've done a a brief test of this using CI 1106 on full naze board. It didn't go all that well. It seems I have absolutely no yaw response at all (PC0, angle & horizon mode). It's as if the yaw channel isn't configured correctly. That made it basically impossible to test the sonar properly I'm afraid.

I've since put 1.8.1 back on, restored my settings and yaw is back.

This was my first CI build test and it's quite possible the fault is with me rather than the code. I'm happy to try things out again some time if needed!

hydra, 2015-05-15:
@Ben250 can you do more testing please?

DIY420, 2015-05-21:
Hi,
I did a test 2 days ago. With dev. build 22th april (last one which stated the issue in the change log). Unfortunately I noticed no improvement. Well, with old SW the copter drifted down slowly, now it drifts up slowly. My config: Baro + sonar activated via control channel (is this correct to activate both?). When I switch from normal mode to alt hold mode, the copter seems to hold some altitude but then drifts up or down slowly. Deadband set to 200, no change. When switching to alt hold mode, the copter makes whooh whooh whooh which seems to be a regulation behaviour (nearby to unstable control loop). So I didn't increase the PID P value any further. Is it true, that I and D do not have any function?
Where can be my mistake? Rest of copter works fine and good performant...

thenickdude, 2015-05-22:
As far as I can see this change didn't fix anything, as it wasn't broken to start with.

>ApplyAltHold is called every loop, which will modify rcCommand[THROTTLE]. However, processRx (which resets rcCommand back to the actual Rx values) is only called at 50 Hz intervals.

`rcCommand` is derived from `rcData` by `annexCode()`, which is always called right before `applyAltHold()`. So applyAltHold never takes its own modified rcCommand values as input.

Fvk1960, 2015-07-25:
I struggled also with althold using sonar. But found sonar can not be used exlusively. so altitude calculated is always influence by baro which by itself is always drifting in y cases.
I do not know howto in cleanflight but in multiwii I made led indicator when altchange is activated.
So I can see when it activated and also easy to get stick back into neutral zone

closed 2017-02-25 by hydra