cleanflight/cleanflight #756: PPM not working in combination with OneShot

JohnieBraaf, 2015-04-11:
I have the following issue on cleanflight 1.8.1

With OneShot enabled PPM is not working.
With OneShot disabled PPM is working.

There is a red LED visible on the Sparky board.

When SBUS is used instead of PPM, channels are working fine with or without OneShot.

ledvinap, 2015-04-11:
It's probably problem with timer sharing. It was solved on CC3D, so applying same fix to sparky should help.

JohnieBraaf, 2015-04-11:
This is probably the commit you are talking about. https://github.com/cleanflight/cleanflight/commit/5096873ab1548a7f5014f3b6d44b297aa04ec4af 
It seems that it checks for TIM_14, that might be a different one on the Sparky.

DaTwo, 2015-04-12:
LOL I was just gonna post this here. Been battling this for a couple hours before figuring out what was causing it to disable PPM. Knew I should have looked here first. :)

JohnieBraaf, 2015-04-12:
Hi DaTwo, it helps that you can confirm the issue. Which is board specific, as I had people on IRC confirming that on Naze it is working fine.

ledvinap, 2015-04-12:
https://github.com/cleanflight/cleanflight/blob/7a4add47d5c1f371586267e153f534840947d22d/src/main/drivers/pwm_mapping.c#L488

It was limited to CC3D to avoid problems elsewhere ...

JohnieBraaf, 2015-04-12:
Tried removing the CC3D condition, but it doesn't work. I'm wondering if TIM_14 is correct, and also what below piece of code does.

void ppmAvoidPWMTimerClash(const timerHardware_t *timerHardwarePtr, TIM_TypeDef *sharedPwmTimer)
{
    if (timerHardwarePtr->tim == sharedPwmTimer) {
        ppmCountShift = 3;  // Divide by 8 if the timer is running at 8 MHz
    }
}

ledvinap, 2015-04-12:
Timer used for oneshot is running on 8MHz, ppmCountShift scales this back to 1MHz that is expected. So you need to set ppmCountShift =3 if PPM timer is shared with OneShot.

hydra, 2015-04-14:
I tested by flashing latest code, enabling PPM and OneShot using a HEX mixer.  I confirm all 6 motor outputs correctly show oneshot signals and the receiver signal is correctly detected.

Test receiver: FrSky D4R-II in 8 channel PPM mode.

@JohnieBraaf Please grab a development build that contains the fix and re-test and confirm the fix also works for you.

closed 2015-04-14 by hydra

JohnieBraaf, 2015-04-14:
@hydra , tested, working like a charm! Thnx a ton :)
Finally able to receive FrSky telemetry over OpenLRS now.
In openLRS, SBUS was occupying the only serial available port...

JohnieBraaf, 2015-04-14:
@DaTwo if you can't wait, placed the dfu file here: https://onedrive.live.com/redir?resid=4416F39ABAB8E8A5%219203

commit 1a2ef223 (2015-04-14): SPARKY - Apply Oneshot/PPM timer clash fix.

Closes #756.