cleanflight/cleanflight #758: Profile 1 and 2 Same Yaw Rate

kitekid99, 2015-04-11:
Hi,

In the Cleanflight GUI 0.63.0, there is a problem with profile 1 and 2. If you change the pitch, roll, or yaw rate in profile 1, and then sets profile 2's pitch, roll, yaw rate to the same as profile 1's. Profile 3 remains unchanged. And vice versa, if you change the pitch, roll, yaw rate in profile 2 it then changes profile 1's to match profile 2. I do not know if this is on purpose or if it has something to do with the rateprofiles but something seems wrong here.

Thanks,
kitekid99

kitekid99, 2015-04-12:
After some more research, Im almost positive it has to do with the rate profiles. It seems profile 1 and profile 2 are somehow using the same rate profile and so when you change the rates in profile 1 it affects profile 2. Not quite sure where to fix this or find it in the code. But, im almost certain it is not supposed to be this way.

straferr, 2015-04-14:
I concur with kitekid99.  It seems the P/R/Yrates, TPA rate values are shared.  Under Cleanflight Configurator GUI, changing the rates from the PID tab for either PID profile 1(0) or 2(1) will affect the other. Profile 3(2) is not affected by this bug and seems fully independent.

hydra, 2015-04-14:
Not a bug. Rate profiles are shared.

The UI needs updating to better reflect this.

Perhaps the documentation needs updating? PR accepted.

closed 2015-04-14 by hydra

kitekid99, 2015-04-15:
Are you saying that profile 1 and 2 have the same rate profile? In reality, that would mean you only really have 2 rate profiles instead of the supposed 3 because 1 and 2 are somehow connected. And I don't think it was meant to be like this. I thought each profile should have it's own rate profile. What's the point of profiles if they don't even change what you want it to change?