cleanflight/cleanflight #763: using the latest ARM GCC compiler (Q1/2015) will generate not working code for the build in USB port

MJ666, 2015-04-13:
Likely not directly connected to the CF code but important to know. Coded generated with latest version of the ARM GCC (Q1/2015) will build code with an non working USB VCP. In general the code is working with this compiler tested on an ALINEWII F1. I was not able to connect anymore to an ALIENWII F3 via USB after upgrading. Any how from LED behavior the reminder of the firmware looks to be working. Windows is reporting an error with the USB device when connecting and the VCP is not showing up. Reverting to the ARM GCC (Q3/2014) version will make it work again with compiling from the same source code. The will likely affect also the Sparky and all other FC's which using the STM USB with VCP. Not sure this will also affect other platforms and operating systems. I'm completely on windows (8.1)

MauroMombelli, 2015-04-27:
arm-none-eabi-gcc (Arch Repository) 4.9.2 20150128 (prerelease) here, working fine for ST32f1XX (Naze) board

MJ666, 2015-04-27:
Naze is not using the build in USB port. It was not affected by the problem.

MJ666, 2015-05-15:
@ledvinap if you working on the USB code this is may be also interesting for you?

hydra, 2015-06-21:
Ok, had a quick look at this, seems the problem is with these flags '-flto -fuse-linker-plugin'

I was able to compile and use USB on the CC3D after disabling them in the Makefile and disabling `#define BLACKBOX` (to free some flash space) and the USB was fine.

hydra, 2015-06-21:
This might have something to do with it:

http://stackoverflow.com/questions/27192018/stm32-usb-unreliable-compilation-with-gcc-os

hydra, 2015-06-22:
Nothing conclusive after a few hours of digging.  Removing `-flto` works round the problem but then the code doesn't fit.  Not a solution by a long shot.

Anyone care to take a look at this since it would be nice to get Cleanflight building using latest GCC for all targets.

nzmichaelh, 2015-06-29:
I've found the problem.  lib/main/STM32_USB-FS-Device_Driver/inc/usb_regs.h has various casts that cast away the volatile from the peripheral registers.  The compiler recognises that there are, say, two reads from the same non-volatile address and optimises away the second read.  This only shows up in LTO mode as LTO mode can optimise across functions.

I'll send a patch.

nzmichaelh, 2015-06-29:
(BTW, I narrowed this down by marking more and more functions with NOINLINE until I found the suspect file)

digitalentity, 2015-06-29:
@nzmichaelh that's a very good news!

closed 2015-07-01 by hydra

hydra, 2015-07-01:
@nzmichaelh I wondered how you tracked it down, thanks for sharing!

tracernz, 2015-07-01:
I think GCC actually has a warning flag to detect when volatile is discarded by a cast? It's in the Atmel ASF SAM4 Makefile because I've encountered it before when compiling their device drivers.

PR #1070 (2015-07-01): usb: mark register accesses as volatile.
Fixes the LTO build with GCC 4.9.  Tested on the STM32F3DISCOVERY.  I
assume the compiler was doing cross function optimisation, saw that
the same non-volatile address was read twice, and dropped the second
read.

Fixes #763.
commit ebc0f53b (2015-07-01): usb: mark register accesses as volatile.

Fixes the LTO build with GCC 4.9.  Tested on the STM32F3DISCOVERY.  I
assume the compiler was doing cross function optimisation, saw that
the same non-volatile address was read twice, and dropped the second
read.

Fixes #763.