cleanflight/cleanflight #770: build fail: missing arm toolchain files

susheelpuranik, 2015-04-14:
1. Being a newbie to Eclipse I installed the whole environment (on Windows 10) with reference to this link 
https://github.com/cleanflight/cleanflight/wiki/Building-with-Eclipse
[Used: Long version method described in the link. Spent a lot of time to verifying that I had followed the instructions perfectly]
2. As per the instructions I downloaded the latest arm gcc toolchain 
https://launchpadlibrarian.net/200702029/gcc-arm-none-eabi-4_9-2015q1-20150306-win32.exe
3. I forked the cleanflight code and tried to build it [no changes were made to it].

Following are the errors that occured:
[image]
In order to investigate the issue I dug into the toolchain folders and the files like 
sys/termios.h 
sys/select.h were actually missing.
Can you please advice on which toolchain and build settings are preferred to compile the forked cleanflight without errors.
Thanks in advance.

wangjun000, 2015-04-14:
so  i just want to say.....the author why  not use keil to creat project？？？

hydra, 2015-04-14:
2014-q2

closed 2015-04-14 by hydra

Redifei, 2015-07-31:
I have same Errors
please help me