cleanflight/cleanflight #771: Motors start when connecting battery

melkati, 2015-04-14:
As soon as I connect the battery the motors start. In my case the four motors start at idle speed. I heard for other only two or three motors and for others they start at full throttle of anything in between.

It happens every single time I apply battery power.

Using:

Flip32+
ESC Afro 12A flashed with BLHeli 13.1 (Oneshot and Damped light active)
Cobra 2204 2300kv motors
Cleanflight 1.8.1

There is my CLI dump:

# dump

# version
Cleanflight/NAZE 1.8.1 Apr 2 2015 / 17:06:11 (988ae2d)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature ONESHOT125
feature BLACKBOX

# map
map AETR1234

# led
led 0 0,0:::0
led 1 0,0:::0
led 2 0,0:::0
led 3 0,0:::0
led 4 0,0:::0
led 5 0,0:::0
led 6 0,0:::0
led 7 0,0:::0
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 0
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set input_filtering_mode = 0
set min_throttle = 1064
set max_throttle = 1864
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 490
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 1
set auto_disarm_delay = 5
set small_angle = 25
set flaps_speed = 0
set fixedwing_althold_dir = 1
set serial_port_1_functions = 1
set serial_port_1_msp_baudrate = 5
set serial_port_1_telemetry_baudrate = 0
set serial_port_1_blackbox_baudrate = 5
set serial_port_1_gps_baudrate = 5
set serial_port_2_functions = 0
set serial_port_2_msp_baudrate = 5
set serial_port_2_telemetry_baudrate = 0
set serial_port_2_blackbox_baudrate = 5
set serial_port_2_gps_baudrate = 5
set serial_port_3_functions = 0
set serial_port_3_msp_baudrate = 5
set serial_port_3_telemetry_baudrate = 0
set serial_port_3_blackbox_baudrate = 5
set serial_port_3_gps_baudrate = 5
set serial_port_4_functions = 0
set serial_port_4_msp_baudrate = 5
set serial_port_4_telemetry_baudrate = 0
set serial_port_4_blackbox_baudrate = 5
set serial_port_4_gps_baudrate = 5
set reboot_character = 82
set gps_provider = 1
set gps_sbas_mode = 1
set gps_auto_config = 1
set gps_auto_baud = 0
set serialrx_provider = 0
set spektrum_sat_bind = 0
set telemetry_switch = 0
set telemetry_inversion = 1
set frsky_default_lattitude = 0.000
set frsky_default_longitude = 0.000
set frsky_coordinates_format = 0
set frsky_unit = 0
set frsky_vfas_precision = 0
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 33
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = 0
set current_meter_type = 1
set align_gyro = 0
set align_acc = 0
set align_mag = 0
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 90
set max_angle_inclination = 500
set gyro_lpf = 42
set moron_threshold = 32
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set yaw_control_direction = 1
set pid_at_min_throttle = 1
set acc_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = 0

# dump profile

# profile
profile 1

# aux
aux 0 1 0 900 1300
aux 1 2 0 1300 1700
aux 2 12 1 1700 2100
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900
aux 20 0 0 900 900
aux 21 0 0 900 900
aux 22 0 0 900 900
aux 23 0 0 900 900
aux 24 0 0 900 900
aux 25 0 0 900 900
aux 26 0 0 900 900
aux 27 0 0 900 900
aux 28 0 0 900 900
aux 29 0 0 900 900
aux 30 0 0 900 900
aux 31 0 0 900 900
aux 32 0 0 900 900
aux 33 0 0 900 900
aux 34 0 0 900 900
aux 35 0 0 900 900
aux 36 0 0 900 900
aux 37 0 0 900 900
aux 38 0 0 900 900
aux 39 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# servo
servo 0 1020 2000 1500 30 -1
servo 1 1020 2000 1500 30 -1
servo 2 1020 2000 1500 100 -1
servo 3 1020 2000 1500 100 -1
servo 4 1020 2000 1500 100 -1
servo 5 1020 2000 1500 100 -1
servo 6 1020 2000 1500 100 -1
servo 7 1020 2000 1500 100 -1

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = 1
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = 1
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_direction = 0
set tri_unarmed_servo = 0
set servo_lowpass_freq = 0
set servo_lowpass_enable = 0
set default_rate_profile = 1
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1000
set failsafe_min_usec = 985
set failsafe_max_usec = 2115
set gimbal_flags = 1
set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff = 5.000
set acc_unarmedcal = 1
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf = 0.600
set baro_cf_vel = 0.985
set baro_cf_alt = 0.965
set mag_declination = 0
set pid_controller = 2
set p_pitch = 30
set i_pitch = 30
set d_pitch = 23
set p_roll = 30
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_pitchf = 2.000
set i_pitchf = 0.600
set d_pitchf = 0.060
set p_rollf = 2.000
set i_rollf = 0.600
set d_rollf = 0.060
set p_yawf = 4.500
set i_yawf = 0.500
set d_yawf = 0.050
set level_horizon = 3.000
set level_angle = 5.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 1

set rc_rate = 95
set rc_expo = 30
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 25
set tpa_rate = 20
set tpa_breakpoint = 1500

nebbian, 2015-04-14:
I've seen lots of people reporting this problem, the common setting seems to be looptime=0.

DominicCliftonNEKTAN, 2015-04-14:
@melkati can you reset your looptime via and re-test?  the default is 3500. 

```
set looptime=3500
save
```

melkati, 2015-04-14:
Setting looptime=3500 fixed the issue, as to the rest of users.

The problem is it's still dangerous, for people that set looptime=0 (or 1000 or 1200 says some people) so if it's not a bug probably the CLI parser should avoid the user set a value lower than X (1200 or whatever).

MJ666, 2015-04-14:
Not sure if minimum looptime should be limited. Setting this to 0 is causing some larger variations in the resulting looptime and can cause all kind of problems depending on the individual settings. I.e. many of the PID controllers are really looptime dependent and are not work well with this variations.

ksal0, 2015-04-14:
PID controller 2, oneshot enabled, looptime = 0
-When I had KISS ESCs installed I had no problems. 
-I switched to blheli flashed ESCs and no other changes then problem started. 
-Set looptime to 1200 or above no issue 
-Set looptime below 1200, two motors spool up to idle after battery connect. 
-Also noticed motors boot up music is still sounding after Naze has finished booting

hydra, 2015-04-14:
Ok, we'll cover this in #546

closed 2015-04-14 by hydra

dkisselev, 2015-04-17:
@ksal0 The boot up music sounding is a blheli feature. They chime once when they initialize, and again when they see a OneShot signal coming from the FC and switch into that mode.

ksal0, 2015-04-18:
Thanks Denis. I realize that's a feature now. And that happens as it should when my looptime is set to 1200 or above. 
But if I set looptime below 1200, motors spin at idle after battery is connected. 
Thats still the main issue.

tiz777, 2016-10-27:
I encountered the same problem with a tiny F3 brushed flight controller,  and to solve it had to:
set motor_pwm_rate=1000 (instead of 400)

Setting looptime higher to 2000 or so, didn't solved the issue.

LindsayFowler, 2017-12-20:
I have two micro SP Racing F3 Evo brushed flight controllers sold on Banggood.com

They don't work with any versions of Cleanflight.
As soon as the battery is connected, all motors hit 7.2V DC.

Setting set motor_pwm_rate=1000 (or any other value) makes no difference.

https://www.banggood.com/Eachine-32bits-F3-Brushed-Flight-Control-Board-Based-On-SP-RACING-F3-EVO-For-Micro-FPV-Frame-p-1076530.html?rmmds=detail-left-hotproducts&cur_warehouse=CN

Here is the CLI dump in case it is of use:

dump
# Cleanflight / SPRACINGF3EVO (SPEV) 2.2.0 Oct 31 2017 / 00:51:40 (782fa33) MSP API: 1.37
name -
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A15
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource LED 1 B08
[... 504 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

#

martinbogo, 2017-12-23:
Same.. I installed Cleanflight 1.2.4 and 2.1 ... both show the same issue with a Banggood Evo F3 -- plug in the battery, and the motors spin at 100% throttle.   

looptime is 1000
min input is 1000, max is 2000, mid is 1500
motor_pwm_rate is 32000

LindsayFowler, 2017-12-25:
Martin, the Bangood SP Racing F3 Evo flight controllers have the same problem when flashed with Betaflight.. it’s probably a problem with the flight controller. Still, a large number of people have reported successful operation. I’d ask for a refund. I have.

martinbogo, 2017-12-25:
No, I won't leave it at that.   

There has to be a reason, and it's worth taking tye time to understand and debug it.  Just saying 'ignore it and ask for a refund' goes against fhe hacker credo.

I'm going to photo the board under a.microscope and.check the chips and orientations of the components.

LindsayFowler, 2017-12-25:
I’m not saying “ignore it”. I like your spirit. I tried probing the board, but I didn’t find any clues. I was hoping to find some blown FETs or some pwm signals. If the chips were on the wrong way, it wouldn’t talk to the USB port during boot load. If the FETs were on the wrong way, you’d see the PWM trains.

mikeller, 2017-12-25:
@martinbogo, @LindsayFowler: Hehe, interesting discussion. Speaking with some experience in the flight controller area, I would say that it's entirely possible that BangGood's supplier borked the first batch of this board with a faulty PCB layout for the motor output drivers, and then fixed it in the second batch - which probably would not stop BangGood from selling the faulty batch.

In an attempt to get this to work, you should try setting `motor_pwm_inversion = on` and see if that helps. If not, I think it's time for a return.

LindsayFowler, 2017-12-26:
Oh okay, that's worth a try! Thanks @mikeller, I'll give it a go.

martinbogo, 2017-12-26:
Same, I wasn't aware you could set motor_pwm_inversion.   Will give it a go.

LindsayFowler, 2017-12-26:
"motor_pwm_inversion = on" is an unknown command in Cleanflight and Betaflight.

mikeller, 2017-12-26:
@LindsayFowler: Have you tried it with `set motor_pwm_inversion = on`?

LindsayFowler, 2017-12-26:
Oh that's right! You need to write "set" when setting values. I forgot. Thank you.

LindsayFowler, 2017-12-26:
Unfortunately, `set motor_pwm_inversion = on` didn't change the output: all motor ports still hit 7.4V DC, as soon as the 2S battery is plugged in. Everything else seems to function properly: the gyros update on screen, the settings are retained when set via USB, boot loader works, there are some  intelligent signals coming off the main chip, but nothing on the inputs to those motor drivers.

martinbogo, 2017-12-26:
Same, I just did:

set motor_pwm_inversion = on, and got 7.4V on M1/M2/M3/M4/M5/M6

here is the config dump:

# dump
# Cleanflight / SPRACINGF3EVO 2.1.0 Jul 30 2017 / 13:13:51 (8b24735)
name -
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A15
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A07
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 A06
resource SERIAL_RX 12 B00
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource LED 1 B08
resource LED 2 NONE
[... 1030 lines cut ...]
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

mikeller, 2017-12-26:
@LindsayFowler, @martinbogo: Looks like your boards are duds, sorry guys (could be that they soldered on the wrong driver FET's or something). I think it's time for a return.

dako76, 2019-10-19:
Have same issu but only one motor (3) is spinning up a short time. I test this with Betaflight, same probleme with diffrent version of firmware. The Brushed board was shipped with Cleanflight/SPRACINGF3EVO 1.13.0 with this everthing was fine. But really old one :-)

leonid-s-usov, 2019-11-04:
I have the same issue with the board bought here https://ru.aliexpress.com/item/32949752878.html?spm=a2g0s.12269583.0.0.ece25a79ovm7KL claimed to be F3 EVO V2
 ```
# version
# Cleanflight/SPRACINGF3EVO 1.13.0 Jun  6 2016 / 00:03:29 (a77bc76)
```
sill hoping to find a solution

martinbogo, 2019-11-04:
I think the PCB design of the clone they are selling has issues ( as stated
by others ) ... not sure if it can be worked around without a board
re-work.  The wrong kind of transistor is being installed on the boards.

leonid-s-usov, 2019-11-04:
I've updated to the latest firmware v 2.5.0 and that has fixed it! Which means that it doesn't seem to be a hardware issue. Maybe a configuration.
I've updated the firmware without erasing the config, and it just worked. I've dumped the config from the stock version so that I can compare and see what's changed

martinbogo, 2019-11-05:
I will try the same.  I haven't uploaded cleanflight 2.5.0 yet.