cleanflight/cleanflight #772: No Tricopter Yaw Servo with low throttle

Redshifft, 2015-04-14:
I can think of reasons why someone did this but I would rather have Rudder control back !!

hydra, 2015-04-14:
try disabling 'pid at min throttle setting' in the cli and report back please.

The bug report isn't very clear, can you give us: a) expected behaviour, b) observed behavior.

a config dump helps too.

Redshifft, 2015-04-14:
:) I just flashed back to 1.7, 
1.8 and 1.81 yaw servo does not function until the throttle is opened....(once I finally figured it out )
I run the sensible "armed motor idle" ;) with a TRI you can still manoever with that thrust, least you could until these versions.
I was interested in the servo "filtering option" with 1,8 .
There is a checkbox called "live servo" in servo Tab, what does that do ?
Still want that dump ?

hydra, 2015-04-14:
This issue, as-reported, is far to confusing to be worked on.  Closing. 

If there is an issue then please describe the problem concisely, the expected behaviour, the observed behaviour and steps to repeat.  Please keep the issue on topic without asking unrelated questions, please also use terminology found in the documentation of cleanflight and provide a config dump.

closed 2015-04-14 by hydra

Redshifft, 2015-04-14:
Thanks for that hydra !
I didn't just post a bug report without spending a few hours trying to resolve and narrow it down.

Redshifft, 2015-04-15:
I did some more testing today at exactly 1100us on throttle yaw control is restored 
found this..
http://www.rcgroups.com/forums/showpost.php?p=31321380&postcount=10591
'pid at min throttle setting' did nothing for me so may be a coincidence.

If you don't get this reply, I will post the bug as I have more detailed information.